Kwiboo/linux-rockchip #2: RK3328 : macphy support is missing

LongChair, 2017-06-08:
Kernel as support for the externel GMACPHY but not internal MACPHY

Kwiboo, 2017-06-09:
We made some initial testing with https://github.com/Kwiboo/linux-rockchip/commit/dac413d854876e5186d3e9ea5eb4d01362e4b5fa but we must have missed something, probably some clock and/or special phy configuration?

SuperDavidWu, 2017-06-16:
I did three patches for the internal phy, which made internal MACPHY worked on my rk3328-evb board.

[0001-net-phy-Add-rockchip-phy-support.txt](https://github.com/Kwiboo/linux-rockchip/files/1080826/0001-net-phy-Add-rockchip-phy-support.txt)
[0002-ethernet-stmmac-dwmac-rk-Add-support-internal-phy-fo.txt](https://github.com/Kwiboo/linux-rockchip/files/1080828/0002-ethernet-stmmac-dwmac-rk-Add-support-internal-phy-fo.txt)
[0003-arm64-dts-rockchip-Add-support-gmac2phy-for-rk3328.txt](https://github.com/Kwiboo/linux-rockchip/files/1080827/0003-arm64-dts-rockchip-Add-support-gmac2phy-for-rk3328.txt)

LongChair, 2017-06-16:
Confirmed working by @kwiboo :)

yanghanxing, 2017-06-20:
We will merge to release-4.4 branch after the code upstream completed.

SuperDavidWu, 2017-06-21:
Hi

Did you have any UI displayed issue, if you apply this patches for internal phy?

LongChair, 2017-06-21:
what kind of UI issues are you talking about ?

SuperDavidWu, 2017-06-21:
Nothing is displayed on the monitor.

LongChair, 2017-06-21:
no i'm using macphy since a few days, i didn't see any display issues

SuperDavidWu, 2017-06-21:
Okay, that‘s all right, thanks！

LongChair, 2017-06-21:
@SuperDavidWu : we were a few commits back compared to GH, i just updated from `13dbe2cc` to `1763310` and now i have a black screen when i boot up.

yanghanxing, 2017-06-23:
I haven't encounter the issue after merge MACPHY patches in lastest kernel code.
@LongChair  please check again for it.

LongChair, 2017-06-25:
Closing this as it's been merged.

closed 2017-06-25 by LongChair