Kwiboo/linux-rockchip #3: RK3328 : USB 2.0 port only works on OTG

LongChair, 2017-06-08:
The USB 2.0 prot doesn't work in host mode, only on OTG

yanghanxing, 2017-06-08:
The patches is offered.

LongChair, 2017-06-08:
Will check the patches and close if fixed :)

LongChair, 2017-06-08:
I applied those patches on top of the 3328 kernel branch and using the 3328-evb dts file.
The USB2 port still doesn't work when i plug a keyboard into it 

if i plug keyboard in USB3 port i get :
```PlexMediaPlayer:~ # lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=ax88179_178a, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
```

plugging in the USB2 port, i get :
```/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=ax88179_178a, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M```

yanghanxing, 2017-06-09:
Did you use arch/arm64/boot/dts/rockchip/rk3328-evb.dts also ?
I suppose you use the rk3328-box.dts right?

yanghanxing, 2017-06-09:
you should remove the dr_mode define in dts file.
&usb20_otg {
--	dr_mode = "peripheral";
	status = "okay";
};

yanghanxing, 2017-06-09:
Due to the hardware design RK3328 otg_id pin no connection out.So usb2.0 otg port can't auto switch host/device mode.

After apply these patches, you can swith usb2.0 otg port（default device mode） to host mode by command.
The command is "echo host > /sys/kernel/debug/ff450000.syscon:usb2-phy@100/mode".

    Usage:
    [1] Force host mode
        echo host > sys/kernel/debug/<u2phy dev name>/mode

    [2] Force peripheral mode
        echo peripheral > sys/kernel/debug/<u2phy dev name>/mode

    [3] Force otg mode
        echo otg > sys/kernel/debug/<u2phy dev name>/mode
    u2phy dev name in RK3328 is "ff450000.syscon:usb2-phy@100"

Another way to set host mode for usb2.0 otg port is below:
Modify the device dts file .
diff --git a/arch/arm64/boot/dts/rockchip/rk3328-evb.dts b/arch/arm64/boot/dts/rockchip/rk3328-evb.dts
index 060cd50..b33f214 100644
--- a/arch/arm64/boot/dts/rockchip/rk3328-evb.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3328-evb.dts
@@ -455,6 +455,7 @@
 };

 &usb20_otg {
+       dr_mode = "host";
        status = "okay";
 };

If you config host mode default,you can't switch to device by commnad.

LongChair, 2017-06-09:
The latter patch works for us, you're good to push it :)

closed 2017-06-09 by LongChair