Kwiboo/linux-rockchip #4: RK3328 : VPU playback issues

LongChair, 2017-06-08:
A few videos tend to show unexpected behaviors. Here are two samples :

- http://demo-uhd3d.com/fiche.php?cat=uhd&id=143  that one will freeze using mpp.
- http://demo-uhd3d.com/fiche.php?cat=uhd&id=145 that one doesn’t play smoothly although it has reasonnable bitrate for VPU2 (60-80 Mbits/s). I can play smoothly other videos which have a significant higher bitrate (120-140 Mbits/s).

Will provide kernel log for each playback

LongChair, 2017-06-14:
The second video stuttering seems to be solved with running VPU @600Mhz.

So we only have the first video issue at this point :)

yanghanxing, 2017-06-15:
@ayaka check both videos , can not find problem

LongChair, 2017-06-19:
We made further investigations.

One the first video, i can get some log from commandline playback :
```
mpi: mpp version: 9758dfa author: Randy Li [osal]: fixup for build in linux
[vd] Selected video codec: hevc (HEVC (High Efficiency Video Coding))
[cplayer] Starting playback...
[cache] Dropping cache at pos 90487, cached range: 1204810740-1204810814.
[cache] Seeking underlying stream: 1204810814 -> 90487
H265D_PARSER: No start code is found.
H265D_PARSER: mpp_hevc_decode_nal_sei error ret = -1006
H265PARSER_PS: vps_id = 0x0
mpp_buf_slot: new width 3840 height 2160 stride hor 4864 ver 2160 fmt    1
[ffmpeg] hevc_rkmpp: Decoder noticed an info change (3840x2160), format=1
H265D_PARSER: IS_IRAP frame found error
H265PARSER_PS: vps_id = 0x0
H265_PARSER_REF: Could not find ref with POC 114
H265_PARSER_REF: generate_missing_ref in
H265_PARSER_REF: generate_missing_ref frame poc 114 slot_index 2
```
@hizukiayaka : hope this can help

LongChair, 2017-06-19:
On the second one what we found out is that is seems that without display the video can be decoded properly at 60 fps. but as soon as we use drm/vop to display it then it can't keep up with rendering.

Could this be a display system issue (maybe VOP) ?

hizukiayaka, 2017-06-19:
Confirmed

yanghanxing, 2017-06-20:
@ayaka , anything update to @LongChair ？

LongChair, 2017-08-26:
Both videos don't seem to be available anymore from the website ... maybe we should close this issue and there is not real way to reproduce them.

Kwiboo, 2017-08-26:
Closing as samples is removed, possibly also fixed by https://github.com/Kwiboo/linux-rockchip/issues/22#issuecomment-324446793

closed 2017-08-26 by Kwiboo