Kwiboo/linux-rockchip #7: RK3328 : GPU performance

LongChair, 2017-06-08:
When using Kodi / PMP on 3328 we can't reach 60 fps even in 1080p.

We tried to overclock the gpu adding two other points to the gpu dvfs (600 Mhz and 700 mhz), but that didn't seem to improve.

also using `glmark2` with or without overlocking didn't seem to change performance.

the sysfs entry `/sys/devices/platform/ff300000.gpu/devfreq/ff300000.gpu/current_freq` would show 7000 mhz, but the clk_summuary seems to show 6000 mhz no matter what :
```
          aclk_gpu_pre                    1            1   600000000          0 0
             aclk_gpu_niu                 0            0   600000000          0 0
             aclk_gpu                     2            1   600000000          0 0
```

yanghanxing, 2017-06-20:
This issue, our engineer think it meet expectation.
Because RK3328 use mali450 mp2 core gpu.
So the glmark2 score should be lower than Amlogic 905.

Kwiboo, 2017-07-30:
Closing since engineer think it meet expectation

closed 2017-07-30 by Kwiboo