Kwiboo/linux-rockchip #8: RK3328 : Missing CEC Support

LongChair, 2017-06-08:
Looks like this feature could be missing, we need to test more

Kwiboo, 2017-06-09:
I plan on testing CEC with a mixture of @mcerveny's [cec-framework backport](https://github.com/mcerveny/rockchip-linux/commits/release-4.4-cec) and @gdachs's [tinker board testing](https://github.com/Kwiboo/linux-rockchip/commits/tinkerboard-4.10-cec) sometime this weekend

gdachs, 2017-06-09:
@Kwiboo: why? do you want to solder the missing connection to hdmi cec?

LongChair, 2017-06-09:
@gdachs : kwiboo refers to 3328 board which has the CEC pins that are connected as well as all the required resistors. This is not tinker :)

Kwiboo, 2017-06-11:
The backported CEC framework seems to work great on my RK3328 box using the [rockchip-4.4-cec](https://github.com/Kwiboo/linux-rockchip/commits/rockchip-4.4-cec) branch.

```
LibreELEC:~ # cec-ctl --playback -S
Driver Info:
        Driver Name                : dwhdmi-rockchip
        Adapter Name               : dw_hdmi
        Capabilities               : 0x00000016
                Logical Addresses
                Transmit
                Remote Control Support
        Driver version             : 4.4.66
        Available Logical Addresses: 4
        Physical Address           : 1.2.0.0
        Logical Address Mask       : 0x0800
        CEC Version                : 2.0
        Vendor ID                  : 0x000c03 (HDMI)
        OSD Name                   : 'Playback'
        Logical Addresses          : 1 (Allow RC Passthrough)

          Logical Address          : 11 (Playback Device 3)
            Primary Device Type    : Playback
            Logical Address Type   : Playback
            All Device Types       : Playback
            RC TV Profile          : None
            Device Features        :
                None

        System Information for device 3 (Tuner 1) from device 11 (Playback Device 3):
                CEC Version                : Tx, OK, Rx, OK, Feature Abort
                Physical Address           : 1.3.0.0
                Primary Device Type        : Tuner
                Vendor ID                  : Tx, OK, Rx, OK, Feature Abort
                OSD Name                   : PlayStation VR
                Power Status               : On
        System Information for device 4 (Playback Device 1) from device 11 (Playback Device 3):
                CEC Version                : 1.3a
                Physical Address           : 1.3.1.0
                Primary Device Type        : Playback
                Vendor ID                  : 0x080046 (Sony)
                OSD Name                   : PlayStation 4
                Power Status               : On
        System Information for device 5 (Audio System) from device 11 (Playback Device 3):
                CEC Version                : 1.4
                Physical Address           : 1.0.0.0
                Primary Device Type        : Audio System
                Vendor ID                  : 0x080046 (Sony)
                OSD Name                   : HT-RT5
                Power Status               : On
        System Information for device 8 (Playback Device 2) from device 11 (Playback Device 3):
                CEC Version                : 1.4
                Physical Address           : 1.0.0.0
                Primary Device Type        : Playback
                Vendor ID                  : 0x080046 (Sony)
                OSD Name                   : HT-RT5
                Power Status               : On

        Topology:

            1.0.0.0: Audio System
            1.0.0.0: Audio System
                1.2.0.0: Playback Device 3
                1.3.0.0: Tuner 1
                    1.3.1.0: Playback Device 1
```

Kwiboo, 2017-06-17:
There is an issue with auto-loading of the rc-cec keymap, dw-hdmi is probably initialized before rc-cec. 
```
[    0.958484] IR keymap rc-cec not found
[    0.958987] Registered IR keymap rc-empty
```

Manually loading the `cec` keymap using `ir-keytable` works after removing `, type: CEC`
```
LibreELEC:~ # head -n1 /usr/lib/udev/rc_keymaps/cec
# table cec, type: CEC
LibreELEC:~ # head -n1 ./cec
# table cec
LibreELEC:~ # ir-keytable -w ./cec
Read cec table
Wrote 97 keycode(s) to driver
```

yanghanxing, 2017-06-20:
We will merge HDMI CEC suported code to release-4.4 from the lastest upstream code.
Please wait a moment, and I will notice you when it completed.

Kwiboo, 2017-07-30:
@yanghanxing that sounds great! I will keep updating the rockchip-4.4-cec branch until you have merged the cec-framework into release-4.4 and I will also keep an eye on http://git.armlinux.org.uk/cgit/linux-arm.git/log/?h=drm-dwhdmi-devel if there is something worth bringing into rockchip-4.4-cec.

LongChair, 2017-08-26:
So CEC seems to work for me when i plug rock64 directly to TV using `cec-ctl --playback`

```
PlexMediaPlayer:~ # cec-ctl --playback
Driver Info:
    Driver Name                : dwhdmi-rockchip
    Adapter Name               : dw_hdmi
    Capabilities               : 0x0000001e
        Logical Addresses
        Transmit
        Passthrough
        Remote Control Support
    Driver version             : 4.4.84
    Available Logical Addresses: 4
    Physical Address           : 2.0.0.0
    Logical Address Mask       : 0x0010
    CEC Version                : 2.0
    Vendor ID                  : 0x000c03 (HDMI)
    OSD Name                   : 'Playback'
    Logical Addresses          : 1 (Allow RC Passthrough)
      Logical Address          : 4 (Playback Device 1)
        Primary Device Type    : Playback
        Logical Address Type   : Playback
        All Device Types       : Playback
        RC TV Profile          : None
        Device Features        :
        None
```

Although when I plug it through AVR it doesn't seem to work and `cec-ctl` gives the following output :

```
cec-ctl --playback
Driver Info:
    Driver Name                : dwhdmi-rockchip
    Adapter Name               : dw_hdmi
    Capabilities               : 0x0000001e
        Logical Addresses
        Transmit
        Passthrough
        Remote Control Support
    Driver version             : 4.4.84
    Available Logical Addresses: 4
    Physical Address           : f.f.f.f
    Logical Address Mask       : 0x0000
    CEC Version                : 2.0
    Vendor ID                  : 0x000c03 (HDMI)
    OSD Name                   : 'Playback'
    Logical Addresses          : 1 (Allow RC Passthrough)
      Logical Address          : Not Allocated
        Primary Device Type    : Playback
        Logical Address Type   : Playback
        All Device Types       : Playback
        RC TV Profile          : None
        Device Features        :
        None
```
By the way using a RPI in the exact same configuration works fine.
@Kwiboo  : any thoughts ?

yanghanxing, 2017-08-28:
@LongChair  Did you tried it on other AVR?

Kwiboo, 2017-09-02:
I have now updated the https://github.com/Kwiboo/linux-rockchip/compare/rockchip-4.4...rockchip-4.4-cec branch with a small adjustment to the `BACKPORT` commit (it will now install `cec.h` and `cec-funcs.h` uapi headers).

There is also two new commits https://github.com/Kwiboo/linux-rockchip/compare/rockchip-4.4-cec%5E%5E...rockchip-4.4-cec that work around the following issues:
- The `IR keymap rc-cec not found` issue
- CEC adapter getting an invalid physical address after a source input change

mcerveny, 2018-01-26:
There is working CEC in rockchip-linux/kernel from [UPSTREAM](https://github.com/rockchip-linux/kernel/search?q=cec&type=Commits). I tested (on ROCK64) and closed rockchip-linux/kernel#12 (see https://github.com/rockchip-linux/kernel/issues/12#issuecomment-360885486).

Kwiboo, 2018-01-26:
@mcerveny thanks, there is still parts missing for a complete backport or cec-framework, we are using https://github.com/Kwiboo/linux-rockchip/compare/release-4.4...rockchip-4.4-cec in LibreELEC to get a complete backport including working rc-cec, ioctl compat and plule8/rainshadow-cec support, it also includes a dw-hdmi fix to read edid on hpd event (as an alternative to RK's solution/hack that also updates drm modes and eld).

olluz, 2018-08-15:
What about the Tinkerboard S ? Any chance to get it working there ?

Kwiboo, 2018-09-01:
@olluz CEC should work on TinkerBoard S using the latest LibreELEC master test builds

olluz, 2018-09-01:
@Kwiboo thanks, do you mean the master branch of LibreELEC ?
And do I need any special defconfig or additional patches to make it work ?
I just built an image and KODI will complain that it can't connect to the remote server. cec-ctl will complain that there is no /dev/cec0 device...
Any help is appreciated.
Thanks again!

Kwiboo, 2018-09-02:
@olluz strange, the master branch of LibreELEC should have an updated devicetree and kernel config with CEC enabled for TinkerBoard, nightly images at https://test.libreelec.tv/ should have working CEC

olluz, 2018-09-03:
@Kwiboo Indeed, with the nightly image CEC works. I wonder why it's not working if I build it myself.
What am I doing wrong in checking out master and building with "PROJECT=Rockchip DEVICE=TinkerBoard ARCH=arm make image" ?
Thanks!

closed 2020-01-01 by Kwiboo