Kwiboo/linux-rockchip #10: RK3328 : Improve the bootloader update time

LongChair, 2017-06-11:
Currently the bootloader update in LE takes quite a while .. 1-2 minutes.

This is probably due to the size of the files we are updating.
@Kwiboo mentionned that we could maybe trim those files as they are padded with zeros.
To be investigated :)

Kwiboo, 2017-06-11:
A possible workaround could be to skip idbloader/u-boot/trust update when `/storage/.update/.nobootloader` or similar exits

LongChair, 2017-06-11:
What would create that .nobootloader ?

Maybe another way could be to have some trust.md5 file that we can compare with current md5 and flash only when they don't match .

This way whatever file that didn't change would be skipped ?

LongChair, 2017-06-14:
Here is a bootloader update summuary

[image]

Don't the update rates look very low ? are we writting to sdcard ? 
If so i don't uderstand how we achieve so lower write rates ... @Kwiboo ? :)

LongChair, 2017-06-14:
Ok so using the patch below to increase the blocksize to `64k`makes it behave a lot better :

```

LongChair, 2017-06-16:
Done, closing :)

closed 2017-06-16 by LongChair