Kwiboo/linux-rockchip #9: RK3328 : Various Clocks improvements

LongChair, 2017-06-11:
Currently with stock device tree it seems that we have the following clocks : 

- CPU will run 1.3 Ghz
- GPU will run 500 Mhz
- VPU not sure 100% but most likely 300 Mhz

As far as i know Cortex A53 is rated to 1.5Ghz
GPU can at least go to 600 Mhz by adding a new opp in the dts file for dvfs.
VPU unsure.

It seems that there are a few "by design" clock limitations which are not related to the max IP clock, but to the clocks configuration design.

@yanghanxing : could you please check what maximum frequncies can be achieved on each ones of these IP ?

yanghanxing, 2017-06-12:
Hi，Lionel：

    CPU Max：1296M。 
    GPU Max：600M。
    VPU Max：500M。

系统产品一部 杨汉兴  
E-mail：yhx@rock-chips.com
手  机：18059046979
电  话：0591-83991906-8580
公  司：福州瑞芯微电子股份有限公司
地  址：福建省福州市铜盘路软件大道89号软件园A区20号楼
 
From: Lionel CHAZALLON
Date: 2017-06-11 15:00
To: Kwiboo/linux-rockchip
CC: Hans Yang; Mention
Subject: [Kwiboo/linux-rockchip] RK3328 : Various Clocks improvements (#9)
Currently with stock device tree it seems that we have the following clocks :
CPU will run 1.3 Ghz
GPU will run 500 Mhz
VPU not sure 100% but most likely 300 Mhz
As far as i know Cortex A53 is rated to 1.5Ghz
GPU can at least go to 600 Mhz by adding a new opp in the dts file for dvfs.
VPU unsure.
It seems that there are a few "by design" clock limitations which are not related to the max IP clock, but to the clocks configuration design.
@yanghanxing : could you please check what maximum frequncies can be achieved on each ones of these IP ?

LongChair, 2017-06-12:
Looking at the clock summuary :

cat /sys/kernel/debug/clk/clk_summary | grep vpu
          aclk_vpu_pre                    0            2   300000000          0 0
             aclk_vpu_niu                 0            0   300000000          0 0
             aclk_vpu                     0            1   300000000          0 0
             hclk_vpu_pre                 0            1    75000000          0 0
                hclk_vpu_niu              0            0    75000000          0 0
                hclk_vpu                  0            1    75000000          0 0

It looks vpu runs 300 Mhz and not 500 Mhz.

Is there any way to change it to 500 mhz ?

Le 12/06/2017 à 08:20, Hans Yang a écrit :
Hi，Lionel：

CPU Max：1296M。
GPU Max：600M。
VPU Max：500M。

系统产品一部 杨汉兴
E-mail：yhx@rock-chips.com<mailto:yhx@rock-chips.com>
手 机：18059046979
电 话：0591-83991906-8580
公 司：福州瑞芯微电子股份有限公司
地 址：福建省福州市铜盘路软件大道89号软件园A区20号楼

From: Lionel CHAZALLON
Date: 2017-06-11 15:00
To: Kwiboo/linux-rockchip
CC: Hans Yang; Mention
Subject: [Kwiboo/linux-rockchip] RK3328 : Various Clocks improvements (#9)
Currently with stock device tree it seems that we have the following clocks :
CPU will run 1.3 Ghz
GPU will run 500 Mhz
VPU not sure 100% but most likely 300 Mhz
As far as i know Cortex A53 is rated to 1.5Ghz
GPU can at least go to 600 Mhz by adding a new opp in the dts file for dvfs.
VPU unsure.
It seems that there are a few "by design" clock limitations which are not related to the max IP clock, but to the clocks configuration design.
@yanghanxing : could you please check what maximum frequncies can be achieved on each ones of these IP ?

yanghanxing, 2017-06-13:
The vpu freq is 300MHz default.You can modify in this file.
./drivers/video/rockchip/vcodec/vcodec_service.c:1443

static void vpu_service_set_freq(struct vpu_service_info *pservice,
                                 struct vpu_reg *reg)
{
        enum VPU_FREQ curr = atomic_read(&pservice->freq_status);

        if (curr == reg->freq)
                return;

        atomic_set(&pservice->freq_status, reg->freq);
        switch (reg->freq) {
        case VPU_FREQ_200M: {
                clk_set_rate(pservice->aclk_vcodec, 200*MHZ);
        } break;
        case VPU_FREQ_266M: {
                clk_set_rate(pservice->aclk_vcodec, 266*MHZ);
        } break;
        case VPU_FREQ_300M: {
                clk_set_rate(pservice->aclk_vcodec, 300*MHZ);
        } break;
        case VPU_FREQ_400M: {
                clk_set_rate(pservice->aclk_vcodec, 400*MHZ);
        } break;
        case VPU_FREQ_500M: {
                clk_set_rate(pservice->aclk_vcodec, 500*MHZ);
        } break;
        case VPU_FREQ_600M: {
                clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
        } break;
        default: {
                unsigned long rate = 300*MHZ;

                if (of_machine_is_compatible("rockchip,rk2928g"))
                        rate = 400*MHZ;

                clk_set_rate(pservice->aclk_vcodec, rate);
        } break;
        }
}

You can change "reg->freq" values to raise vpu freq.

LongChair, 2017-06-14:
@yanghanxing :  OK i tried to alter the code above with the following patch to make sure that all cases would run @600 mhz :
```
--- a/drivers/video/rockchip/vcodec/vcodec_service.c	2017-06-14 04:12:38.425268502 -0700
+++ b/drivers/video/rockchip/vcodec/vcodec_service.c	2017-06-14 04:13:14.703156183 -0700
@@ -1450,19 +1450,19 @@
 	atomic_set(&pservice->freq_status, reg->freq);
 	switch (reg->freq) {
 	case VPU_FREQ_200M: {
-		clk_set_rate(pservice->aclk_vcodec, 200*MHZ);
+		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
 	} break;
 	case VPU_FREQ_266M: {
-		clk_set_rate(pservice->aclk_vcodec, 266*MHZ);
+		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
 	} break;
 	case VPU_FREQ_300M: {
-		clk_set_rate(pservice->aclk_vcodec, 300*MHZ);
+		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
 	} break;
 	case VPU_FREQ_400M: {
-		clk_set_rate(pservice->aclk_vcodec, 400*MHZ);
+		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
 	} break;
 	case VPU_FREQ_500M: {
-		clk_set_rate(pservice->aclk_vcodec, 500*MHZ);
+		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
 	} break;
 	case VPU_FREQ_600M: {
 		clk_set_rate(pservice->aclk_vcodec, 600*MHZ);
@@ -1471,7 +1471,7 @@
 		unsigned long rate = 600*MHZ;

 		if (of_machine_is_compatible("rockchip,rk2928g"))
-			rate = 400*MHZ;
+			rate = 600*MHZ;

 		clk_set_rate(pservice->aclk_vcodec, rate);
 	} break;
```

Despite that patch when looking at the vpu freqs, it seems it would run still @300 Mhz 

```
PlexMediaPlayer:~ # cat /sys/kernel/debug/clk/clk_summary | grep vpu
          aclk_vpu_pre                    0            2   300000000          0 0
             aclk_vpu_niu                 0            0   300000000          0 0
             aclk_vpu                     0            2   300000000          0 0
             hclk_vpu_pre                 0            1    75000000          0 0
                hclk_vpu_niu              0            0    75000000          0 0
                hclk_vpu                  0            2    75000000          0 0
```

I add to this the full `clk_summuary`
[clk_summuary.txt](https://github.com/Kwiboo/linux-rockchip/files/1074415/clk_summuary.txt)

omegamoon, 2017-06-14:
Following patch sets the VPU speed at 600MHz on default:

```
--- a/rk3328.dtsi	2017-06-14 15:00:39.203665073 +0200
+++ b/rk3328.dtsi	2017-06-14 15:01:02.747496124 +0200
@@ -881,7 +881,7 @@
 			<300000000>, <100000000>,
 			<300000000>, <200000000>,
 			<400000000>, <500000000>,
-			<200000000>, <300000000>,
+			<200000000>, <600000000>,
 			<300000000>, <250000000>,
 			<200000000>, <100000000>,
 			<24000000>, <100000000>,
```
resulting in following:
```
LibreELEC:~ # cat /sys/kernel/debug/clk/clk_summary | grep vpu
          aclk_vpu_pre                    0            2   600000000          0 0  
             aclk_vpu_niu                 0            0   600000000          0 0  
             aclk_vpu                     0            1   600000000          0 0  
             hclk_vpu_pre                 0            1   150000000          0 0  
                hclk_vpu_niu              0            0   150000000          0 0  
                hclk_vpu                  0            1   150000000          0 0  
```

LongChair, 2017-06-14:
@Kwiboo : might be one patch we want in as well :)

omegamoon, 2017-06-18:
Following patch sets the CPU speed at 1.5GHz and GPU speed at 750MHz
This results in a glmark2 score of 106

```
--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi	2017-06-17 11:27:55.000000000 +0200
+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi	2017-06-18 18:16:15.674252696 +0200
@@ -140,6 +140,36 @@
 			opp-microvolt = <1300000>;
 			clock-latency-ns = <40000>;
 		};
+		opp-1392000000 {
+			opp-hz = /bits/ 64 <1392000000>;
+			opp-microvolt = <1300000>;
+			clock-latency-ns = <40000>;
+		};
+		opp-1416000000 {
+			opp-hz = /bits/ 64 <1416000000>;
+			opp-microvolt = <1350000>;
+			clock-latency-ns = <40000>;
+		};
+		opp-1440000000 {
+			opp-hz = /bits/ 64 <1440000000>;
+			opp-microvolt = <1380000>;
+			clock-latency-ns = <40000>;
+		};
+		opp-1464000000 {
+			opp-hz = /bits/ 64 <1464000000>;
+			opp-microvolt = <1380000>;
+			clock-latency-ns = <40000>;
+		};
+		opp-1488000000 {
+			opp-hz = /bits/ 64 <1488000000>;
+			opp-microvolt = <1390000>;
+			clock-latency-ns = <40000>;
+		};
+		opp-1500000000 {
+			opp-hz = /bits/ 64 <1500000000>;
+			opp-microvolt = <1390000>;
+			clock-latency-ns = <40000>;
+		};
 	};
 
 	arm-pmu {
@@ -626,6 +656,22 @@
 			opp-hz = /bits/ 64 <600000000>;
 			opp-microvolt = <1150000>;
 		};
+		opp-675000000 {
+			opp-hz = /bits/ 64 <675000000>;
+			opp-microvolt = <1200000>;
+		};
+		opp-700000000 {
+			opp-hz = /bits/ 64 <700000000>;
+			opp-microvolt = <1300000>;
+		};
+		opp-720000000 {
+			opp-hz = /bits/ 64 <720000000>;
+			opp-microvolt = <1300000>;
+		};
+		opp-754000000 {
+			opp-hz = /bits/ 64 <754000000>;
+			opp-microvolt = <1350000>;
+		};
 	};
 
 	vpu_service: vpu-service@ff350000 {
@@ -880,7 +926,7 @@
 			<50000000>, <50000000>,
 			<50000000>, <50000000>,
 			<24000000>, <600000000>,
-			<491520000>, <1200000000>,
+			<75400000>, <1500000000>,
 			<150000000>, <75000000>,
 			<75000000>, <150000000>,
 			<75000000>, <75000000>,
```

yanghanxing, 2017-06-20:
The vpu freq you can modify by yourselves.

LongChair, 2017-06-20:
Yeah we have tried to modify the vpu freq by ourselves :
```
cat /sys/kernel/debug/clk/clk_summary | grep vpu
          aclk_vpu_pre                    0            2   600000000          0 0
             aclk_vpu_niu                 0            0   600000000          0 0
             aclk_vpu                     0            2   600000000          0 0
             hclk_vpu_pre                 0            1   150000000          0 0
                hclk_vpu_niu              0            0   150000000          0 0
                hclk_vpu                  0            2   150000000          0 0
```

That didn't seem to improve the vpu issues we were having :)

yanghanxing, 2017-06-20:
You mean the video playback issue what @ayaka confirm now?

LongChair, 2017-06-20:
Yeah ayaka confirmes the issue we had with playing some videos which is one issue.
Then we still have some performance issue on files like LG chess, not sure if this one was confirmed.

LongChair, 2017-06-25:
I will close that one for now, overclocking is not a priority, we want to get it stable first, and find out eventual issues with using the hardware normally before trying to bump frequencies, should we need it.

closed 2017-06-25 by LongChair