Kwiboo/linux-rockchip #11: RK3328: HDMI issues

yanghanxing, 2017-06-13:
1.if i plug to the 1080p AVR -> 4KTV monitor -> (get only a 800x600 resolution)
2.if i plug to the 1080p AVR -> 1080p Monitor -> ( have a black screen, no signal )

yanghanxing, 2017-06-13:
The first problem: We can't find on our box. Please make sure HDMI connected when power on , Is it only one resultion in display setting?

The second problem: We can provide a patch to you .Please try and reply result to me.

yanghanxing, 2017-06-13:
diff --git a/drivers/gpu/drm/bridge/dw-hdmi.c b/drivers/gpu/drm/bridge/dw-hdmi.c
index 912210b..fddb006 100644
--- a/drivers/gpu/drm/bridge/dw-hdmi.c
+++ b/drivers/gpu/drm/bridge/dw-hdmi.c
@@ -1663,9 +1663,9 @@ static void hdmi_av_composer(struct dw_hdmi *hdmi,
 		HDMI_FC_INVIDCONF_IN_I_P_INTERLACED :
 		HDMI_FC_INVIDCONF_IN_I_P_PROGRESSIVE;
 
-	inv_val |= hdmi->sink_is_hdmi ?
-		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
-		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
+//	inv_val |= hdmi->sink_is_hdmi ?
+//		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
+//		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
 
 	hdmi_writeb(hdmi, inv_val, HDMI_FC_INVIDCONF);
 
@@ -1925,7 +1925,9 @@ static int dw_hdmi_setup(struct dw_hdmi *hdmi, struct drm_display_mode *mode)
 	/* not for DVI mode */
 	if (hdmi->sink_is_hdmi) {
 		dev_dbg(hdmi->dev, "%s HDMI mode\n", __func__);
-
+		hdmi_modb(hdmi, HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF);
 		/* HDMI Initialization Step F - Configure AVI InfoFrame */
 		hdmi_config_AVI(hdmi, mode);
 		hdmi_config_vendor_specific_infoframe(hdmi, mode);
diff --git a/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c b/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
index 8c67659..8d48d70 100644
--- a/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
+++ b/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
@@ -254,7 +254,7 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		return -EINVAL;
 
 	/* set pdata_en to 0 */
-	inno_phy_modeb(hdmi, 0, 1, 0x01);
+	inno_phy_modeb(hdmi, 0, 1, 0x02);
 	/* Power off post PLL */
 	inno_phy_modeb(hdmi, 1, 1, 0xaa);
 
@@ -273,14 +273,6 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		inno_phy_writel(hdmi, 0xab, val);
 		inno_phy_writel(hdmi, 0xaa, 0x0e);
 	}
-	/* Power up post PLL */
-	inno_phy_modeb(hdmi, 0, 1, 0xaa);
-	/* Wait for post PLL lock */
-	for (i = 0; i < 5; ++i) {
-		if (inno_phy_readl(hdmi, 0xaf) & 1)
-			break;
-		usleep_range(1000, 2000);
-	}
 
 	for (i = 0; i < 14; i++)
 		inno_phy_writel(hdmi, 0xb5 + i, inno_phy_config->regs[i]);
@@ -317,13 +309,26 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		inno_phy_writel(hdmi, 0xc5, 0x81);
 	}
 
-	if (inno_phy_config->tmdsclock > 340000000)
-		msleep(100);
-
+	/* Power up post PLL */
+	inno_phy_modeb(hdmi, 0, 1, 0xaa);
+	/* Power up tmds driver */
 	inno_phy_modeb(hdmi, 4, 4, 0xb0);
 	inno_phy_writel(hdmi, 0xb2, 0x0f);
+
+	/* Wait for post PLL lock */
+	for (i = 0; i < 5; ++i) {
+		if (inno_phy_readl(hdmi, 0xaf) & 1)
+			break;
+		usleep_range(1000, 2000);
+	}
+	if (!(inno_phy_readl(hdmi, 0xaf) & 1)) {
+		dev_err(hdmi->dev, "HDMI PHY Post PLL unlock\n");
+		return -ETIMEDOUT;
+	}
+	if (inno_phy_config->tmdsclock > 340000000)
+		msleep(100);
 	/* set pdata_en to 1 */
-	inno_phy_modeb(hdmi, 1, 1, 0x01);
+	inno_phy_modeb(hdmi, 1, 1, 0x02);
 	return 0;
 }

yanghanxing, 2017-06-13:
`diff --git a/drivers/gpu/drm/bridge/dw-hdmi.c b/drivers/gpu/drm/bridge/dw-hdmi.c
index 912210b..fddb006 100644
--- a/drivers/gpu/drm/bridge/dw-hdmi.c
+++ b/drivers/gpu/drm/bridge/dw-hdmi.c
@@ -1663,9 +1663,9 @@ static void hdmi_av_composer(struct dw_hdmi *hdmi,
 		HDMI_FC_INVIDCONF_IN_I_P_INTERLACED :
 		HDMI_FC_INVIDCONF_IN_I_P_PROGRESSIVE;
 
-	inv_val |= hdmi->sink_is_hdmi ?
-		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
-		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
+//	inv_val |= hdmi->sink_is_hdmi ?
+//		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
+//		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
 
 	hdmi_writeb(hdmi, inv_val, HDMI_FC_INVIDCONF);
 
@@ -1925,7 +1925,9 @@ static int dw_hdmi_setup(struct dw_hdmi *hdmi, struct drm_display_mode *mode)
 	/* not for DVI mode */
 	if (hdmi->sink_is_hdmi) {
 		dev_dbg(hdmi->dev, "%s HDMI mode\n", __func__);
-
+		hdmi_modb(hdmi, HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF);
 		/* HDMI Initialization Step F - Configure AVI InfoFrame */
 		hdmi_config_AVI(hdmi, mode);
 		hdmi_config_vendor_specific_infoframe(hdmi, mode);
diff --git a/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c b/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
index 8c67659..8d48d70 100644
--- a/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
+++ b/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c
@@ -254,7 +254,7 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		return -EINVAL;
 
 	/* set pdata_en to 0 */
-	inno_phy_modeb(hdmi, 0, 1, 0x01);
+	inno_phy_modeb(hdmi, 0, 1, 0x02);
 	/* Power off post PLL */
 	inno_phy_modeb(hdmi, 1, 1, 0xaa);
 
@@ -273,14 +273,6 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		inno_phy_writel(hdmi, 0xab, val);
 		inno_phy_writel(hdmi, 0xaa, 0x0e);
 	}
-	/* Power up post PLL */
-	inno_phy_modeb(hdmi, 0, 1, 0xaa);
-	/* Wait for post PLL lock */
-	for (i = 0; i < 5; ++i) {
-		if (inno_phy_readl(hdmi, 0xaf) & 1)
-			break;
-		usleep_range(1000, 2000);
-	}
 
 	for (i = 0; i < 14; i++)
 		inno_phy_writel(hdmi, 0xb5 + i, inno_phy_config->regs[i]);
@@ -317,13 +309,26 @@ inno_dw_hdmi_phy_init(struct dw_hdmi *dw_hdmi, void *data,
 		inno_phy_writel(hdmi, 0xc5, 0x81);
 	}
 
-	if (inno_phy_config->tmdsclock > 340000000)
-		msleep(100);
-
+	/* Power up post PLL */
+	inno_phy_modeb(hdmi, 0, 1, 0xaa);
+	/* Power up tmds driver */
 	inno_phy_modeb(hdmi, 4, 4, 0xb0);
 	inno_phy_writel(hdmi, 0xb2, 0x0f);
+
+	/* Wait for post PLL lock */
+	for (i = 0; i < 5; ++i) {
+		if (inno_phy_readl(hdmi, 0xaf) & 1)
+			break;
+		usleep_range(1000, 2000);
+	}
+	if (!(inno_phy_readl(hdmi, 0xaf) & 1)) {
+		dev_err(hdmi->dev, "HDMI PHY Post PLL unlock\n");
+		return -ETIMEDOUT;
+	}
+	if (inno_phy_config->tmdsclock > 340000000)
+		msleep(100);
 	/* set pdata_en to 1 */
-	inno_phy_modeb(hdmi, 1, 1, 0x01);
+	inno_phy_modeb(hdmi, 1, 1, 0x02);
 	return 0;
 }
 
`

yanghanxing, 2017-06-13:
[3328_hdmi.txt](https://github.com/Kwiboo/linux-rockchip/files/1070192/3328_hdmi.txt)

yanghanxing, 2017-06-13:
1.if i plug to the 1080p AVR -> 4KTV monitor -> (get only a 800x600 resolution)
this problem ,pls try this patch below

arch/arm64/boot/dts/rockchip/rk3328-evb.dts
&hdmi {
	#sound-dai-cells = <0>;
+	ddc-i2c-scl-high-time-ns = <9625>;
+	ddc-i2c-scl-low-time-ns = <10000>;
	status = "okay";
};

LongChair, 2017-06-14:
Applies both patches :

1. i still get only one mode available 
```
PlexMediaPlayer:~ # cat /sys/class/drm/card0-HDMI-A-1/mode
800x600p60
```

2. still getting a black screen

yanghanxing, 2017-06-14:
please send the each log to us

系统产品一部 杨汉兴  
E-mail：yhx@rock-chips.com
手  机：18059046979
电  话：0591-83991906-8580
公  司：福州瑞芯微电子股份有限公司
地  址：福建省福州市铜盘路软件大道89号软件园A区20号楼
 
From: Lionel CHAZALLON
Date: 2017-06-14 13:05
To: Kwiboo/linux-rockchip
CC: Hans Yang; Author
Subject: Re: [Kwiboo/linux-rockchip] RK3328: HDMI issues (#11)
Applies both patches :
i still get only one mode available
PlexMediaPlayer:~ # cat /sys/class/drm/card0-HDMI-A-1/mode
800x600p60

still getting a black screen

LongChair, 2017-06-14:
Here are the logs :

[log.for.1.txt](https://github.com/Kwiboo/linux-rockchip/files/1073607/log.for.1.txt)
[log.for.2.txt](https://github.com/Kwiboo/linux-rockchip/files/1073608/log.for.2.txt)

Also on both tests i checked :
```
PlexMediaPlayer:~ # cat /sys/class/drm/card0-HDMI-A-1/mode
800x600p60
```
I am not 100% sure that the 1080p monitor supports 800x600, so could be that it can't use that mode on the 1080p screen and then gives a black screen.

yzheng2012, 2017-06-15:
Hi
1. What is the model of AVR？
2. Is HDMI okay if RK3328 is direct connected to TV/monitor?

Please use attachment patch test, In this patch, i add some edid error log, and list common HDMI mode if reading edid is failed. Please test and send us the kernel log.
[2.txt](https://github.com/Kwiboo/linux-rockchip/files/1076483/2.txt)

LongChair, 2017-06-15:
Hi @yzheng2012 

1. AVR is a Denon AVR-1910.
2. Yes when connected directly to the TV it works fine  both with 1080p Monitor and 4K TV. Using several devices through this AVR (Raspberry Pi, Wetek Hub) work fine.

Here is a log with your patch applied. Was used with AVR + 4KTV.
[kernel.avr.4Ktv.txt](https://github.com/Kwiboo/linux-rockchip/files/1076543/kernel.avr.4Ktv.txt)

LongChair, 2017-06-15:
@yzheng2012, I updated the logfile above, make sure you use the right one :)

yzheng2012, 2017-06-15:
@LongChair, I can find any edid reading in the log ,such as Please"failed to get edid".
Please update log file again.
With this patch, AVR + 4K TV, How many mode in the modes node?
cat /sys/class/drm/card0-HDMI-A-1/mod

LongChair, 2017-06-15:
Here is the log again with RK3288 box -> AVR -> 4KTV 

[kernel.again.txt](https://github.com/Kwiboo/linux-rockchip/files/1076724/kernel.again.txt)

it still lists only one mode :
```
PlexMediaPlayer:~ # cat /sys/class/drm/card0-HDMI-A-1/modes
800x600p60
```

yzheng2012, 2017-06-15:
Is the patch properly patched? I can not find any log I added.
The right log should have following message:
[    0.455686] dw_hdmi_i2c_read length 1
[    0.456863] dw_hdmi_i2c_read length 128
[    0.561389] dw_hdmi_i2c_read length 128
[    3.814426] dw_hdmi_i2c_read length 1
[    3.821079] dw_hdmi_i2c_read length 128
[    3.955850] dw_hdmi_i2c_read length 128

LongChair, 2017-06-15:
yeah i checked the kernel code to see if patch was applied and it is.
I suppose that if we don't get those logs, it means that we don't get up to this code :)

LongChair, 2017-06-16:
@yzheng2012 @yanghanxing : I rebuild my full LE tree and now both of the dts patches & code one above seem to fix the issue.

apologies for the misleading wrong testing.

@yanghanxing : can this be merged to `release-4.4`branch ? :)

LongChair, 2017-06-19:
@yzheng2012 , by the way the issue https://github.com/Kwiboo/linux-rockchip/issues/15 is related to HDMI system as well. could you have a look ? :)

yanghanxing, 2017-06-20:
@yzheng2012 Please help @LongChair confirm this problem.

When I connect my RK3328 box to the 4K display i have which supports 4K@60 modes, i can't seem to get that mode.

The following modes will be listed :

cat /sys/class/drm/card0-HDMI-A-1/modes
4096x2160p30
4096x2160p30
4096x2160p25
4096x2160p24
4096x2160p24
3840x2160p30
3840x2160p30
3840x2160p25
3840x2160p24
3840x2160p24
1920x1080p60
1920x1080p60
1920x1080i60
1920x1080i60
1920x1080p50
1920x1080i50
1920x1080i48
1920x1080p30
1920x1080p30
1920x1080p25
1920x1080p24
1920x1080p24
1280x720p60
1280x720p60
1280x720p50
800x600p60
720x576p50
720x576i50
720x480p60
720x480i60
I grabbed the edid from :

hexdump /sys/class/drm/card0-HDMI-A-1/edid
0000000 ff00 ffff ffff 00ff b358 3700 0000 0000
0000010 1901 0301 5f80 7836 cf0a a374 4c57 23b0
0000020 4809 af4c 80ef 00b3 0095 40a9 4090 0081
0000030 8081 4081 0101 e808 3000 70f2 805a 58b0
0000040 008a 1d50 0074 1e00 3a02 1880 3871 402d
0000050 2c58 0045 1d50 0074 1e00 0000 fc00 3400
0000060 5533 4448 4c5f 4443 545f 0a56 0000 fd00
0000070 3000 0f3e 3c46 0a00 2020 2020 2020 c301
0000080 0302 f147 015c 0706 0302 1615 1211 0413
0000090 0514 901f 2120 5d22 5f5e 6160 6362 6564
00000a0 2c66 0709 1507 5007 063f 57c0 0006 0183
00000b0 0000 036e 000c 0020 3cf8 0020 0480 0203
00000c0 e501 000f 6000 010c 801d 733e 2d38 7e40
00000d0 452c 0080 52d0 0000 011e 801d 72d0 2d1c
00000e0 1020 252c 0080 52d0 0000 009e 0000 0000
00000f0 0000 0000 0000 0000 0000 0000 0000 f800
0000100

LongChair, 2017-06-20:
For the former issue it's fine, we consider it fiex, just waiting for it ti be merged to Github. will clos it once it's done :)

The 4K@60 fps issue is still there but i made another issue for it https://github.com/Kwiboo/linux-rockchip/issues/15

LongChair, 2017-06-25:
closing that one as it's been merged https://github.com/Kwiboo/linux-rockchip/issues/15 is still an issue, but lets handle it there.

closed 2017-06-25 by LongChair

yanghanxing, 2017-06-26:
@yzheng2012 Please confirm this issue:
https://github.com/Kwiboo/linux-rockchip/issues/15