Kwiboo/linux-rockchip #12: RK3328 : Decoding issues

LongChair, 2017-06-14:
Since the 4.4 kernel merge and updating to it, as well updating mpp, we have various playback crashes.

Some files will play about right, some other result in a black screen.

Checking kernel log will show :

```
<6>[   26.587045] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
<6>[   29.274911] ax88179_178a 1-1.4:1.0 eth0: ax88179 - Link status is: 1
<6>[   29.281958] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<4>[   30.312222] rk_iommu ff373f00.iommu: Unbalanced pm_runtime_enable!
<6>[   31.855184] drm_crtc_check_viewport: skip check
<3>[   67.891443] rk-vcodec ff360000.rkvdec: can not find 123 buffer in list
<4>[   67.891772] rk_iommu ff360480.iommu: Unbalanced pm_runtime_enable!
<3>[   67.911178] dma-pl330 ff1f0000.dmac: fill_queue:2111 Bad Desc(2)
```

and then after a few secs we get 

```
[  277.551435] rk-vcodec ff9a0000.vpu-service: closed
[  277.565007] rk_iommu ff930300.iommu: Page fault at 0x024adc80 of type read
[  277.565020] rk_iommu ff930300.iommu: iova = 0x024adc80: dte_index: 0x9 pte_index: 0xad page_offset: 0xc80
[  277.565029] rk_iommu ff930300.iommu: mmu_dte_addr: 0x2e1d2000 dte@0x2e1d2024: 0x21762001 valid: 1 pte@0x217622b4: 0x1f3c5006 valid: 0 page@0x00000000 flags: 0x0
[  277.565034] VOP [ff940000.vop]: DISABLED
[  277.565038]
[  277.565038] 0x00000000: 00000001 03007236 00803000 00000000
[  277.565038] 0x00000010: 000000bf 0000e401 00000000 00711c08
[  277.565038] 0x00000020: 00461020 00000000 00000000 00000000
[  277.565038] 0x00000030: 00000029 45400000 00000000 01e001e0
[  277.565038] 0x00000040: 023cf000 0254d800 0327077f 0327077f
[  277.565038] 0x00000050: 00b100c0 10001000 7fae7fdd 00000000
[  277.565038] 0x00000060: 00000000 00000000 00000000 00000000
[  277.565038] 0x00000070: 00000081 00400000 00000000 01400780
[  277.565038] 0x00000080: 00fd2000 00000000 0437077f 0437077f
[  277.565038] 0x00000090: 002900c0 10001000 10001000 00000000
[  277.565038] 0x000000a0: 00000069 000000c0 00000000 00000000
[  277.565038] 0x000000b0: 00000001 00000000 01400140 01400140
[  277.565038] 0x000000c0: 00000000 00ef013f 000a000a 00000000
[  277.565038] 0x000000d0: 00000000 00ef013f 000a000a 00000000
00000001 00000000 01400140 01400140
[  277.565167] 0x00000110: 00000000 00ef013f 000a000a 00000000
[  277.565167] 0x00000120: 00000000 00ef013f 000a000a 00000000
[  277.565167] 0x00000130: 00000000 00ef013f 000a000a 00000000
[  277.565167] 0x00000140: 00000000 00ef013f 000a000a 00000000
[  277.565167] 0x00000150: 00000000 00000000 00000000 000a000a
[  277.565167] 0x00000160: 00000000 00000000 00000000 00000000
[  277.565167] 0x00000170: 00c00840 00290461 10001000 00000000
[  277.565167] 0x00000180: 00000003 000a00fa 0abe002c 00c00840
[  277.565167] 0x00000190: 04650005 00290461 00000000 VOP [ff930000.vop]: ACTIVE
[  277.565243]     Connector: HDMI-A
[  277.565247]  bus_format[100a] output_mode[f]
[  277.565251]     Display mode: 1920x1080p24
[  277.565255]  clk[74250] real_clk[74250] type[40] flag[5]
[  277.565259]  H: 1920 2558 2602 2750
[  277.565262]  V: 1080 1084 1089 1125
[  277.565265]     win0-0: DISABLED
[  277.565269]     win1-0: ACTIVE
[  277.565274]  format: AR24 little-endian (0x34325241)
[  277.565276]  zpos: 1
[  277.565279]  src: pos[0x0] rect[1920x1080]
[  277.565283]  dst: pos[0x0] rect[1920x1080]
[  277.565288]  buf[0]: addr: 0x00fd2000 pitch: 7680 offset: 0
[  277.565292]     win2-0: DISABLED
[  277.565296]     win2-0: DISABLED
[  277.565299]     win2-1: DISABLED
[  277.565302]     win2-2: DISABLED
[  277.565305]     win3-0: DISABLED
[  277.565308]     win3-0: DISABLED
[  277.565312]     win3-1: DISABLED
[  277.565315]     win3-2: DISABLED
[  277.565352] rk_iommu ff930300.iommu: Page fault at 0x024adc80 of type read
[  277.565360] rk_iommu ff930300.iommu: iova = 0x024adc80: dte_index: 0x9 pte_index: 0xad page_offset: 0xc80
[  277.565368] rk_iommu ff930300.iommu: mmu_dte_addr: 0x2e1d2000 dte@0x2e1d2024: 0x21762001 valid: 1 pte@0x217622b4: 0x1f3c5006 valid: 0 page@0x00000000 flags: 0x0
[  277.565372] VOP [ff940000.vop]: DISABLED
[  277.565375]
[  277.565375] 0x00000000: 00000001 03007236 00803000 00000000
[  277.565375] 0x00000010: 000000bf 0000e401 00000000 00711c08
[  277.565375] 0x00000020: 00461020 00000000 00000000 00000000
[  277.565375] 0x00000030: 00000029 45400000 00000000 01e001e0
[  277.565375] 0x00000040: 023cf000 0254d800 0327077f 0327077f
[  277.565375] 0x00000050: 00b100c0 10001000 7fae7fdd 00000000
[  277.565375] 0x00000060: 00000000 00000000 00000000 00000000
[  277.565375] 0x00000070: 00000081 00400000 00000000 01400780
[  277.565375] 0x00000080: 00fd2000 00000000 0437077f 0437077f
[  277.565375] 0x00000090: 002900c0 10001000 10001000 00000000
[  277.565375] 0x000000a0: 00000069 000000c0 00000000 00000000
[  277.565375] 0x000000b0: 00000001 00000000 01400140 01400140
[  277.565375] 0x000000c0: 00000000 00ef013f 000a000a 00000000
[  277.565375] 0x000000d0: 00000000 00ef013f 000a000a 00000000
00000001 00000000 01400140 01400140
[  277.565497] 0x00000110: 00000000 00ef013f 000a000a 00000000
[  277.565497] 0x00000120: 00000000 00ef013f 000a000a 00000000
[  277.565497] 0x00000130: 00000000 00ef013f 000a000a 00000000
[  277.565497] 0x00000140: 00000000 00ef013f 000a000a 00000000
[  277.565497] 0x00000150: 00000000 00000000 00000000 000a000a
[  277.565497] 0x00000160: 00000000 00000000 00000000 00000000
[  277.565497] 0x00000170: 00c00840 00290461 10001000 00000000
[  277.565497] 0x00000180: 00000003 000a00fa 0abe002c 00c00840
[  277.565497] 0x00000190: 04650005 00290461 00000000 VOP [ff930000.vop]: ACTIVE
[  277.565571]     Connector: HDMI-A
[  277.565574]  bus_format[100a] output_mode[f]
[  277.565577]     Display mode: 1920x1080p24
[  277.565581]  clk[74250] real_clk[74250] type[40] flag[5]
[  277.565584]  H: 1920 2558 2602 2750
[  277.565587]  V: 1080 1084 1089 1125
[  277.565590]     win0-0: DISABLED
[  277.565593]     win1-0: ACTIVE
[  277.565597]  format: AR24 little-endian (0x34325241)
[  277.565600]  zpos: 1
[  277.565603]  src: pos[0x0] rect[1920x1080]
[  277.565606]  dst: pos[0x0] rect[1920x1080]
[  277.565610]  buf[0]: addr: 0x00fd2000 pitch: 7680 offset: 0
[  277.565613]     win2-0: DISABLED
[  277.565616]     win2-0: DISABLED
[  277.565619]     win2-1: DISABLED
[  277.565622]     win2-2: DISABLED
[  277.565625]     win3-0: DISABLED
[  277.565628]     win3-0: DISABLED
[  277.565631]     win3-1: DISABLED
[  277.565634]     win3-2: DISABLED
[  277.573043] drm_crtc_check_viewport: skip check
[  277.591894] rk_iommu ff930300.iommu: Enable stall request timed out, status: 0x00004b
[  277.601895] rk_iommu ff930300.iommu: Disable paging request timed out, status: 0x00008b
[  277.665255] rk_iommu ff930300.iommu: Enable stall request timed out, status: 0x00008b
[  277.665263] rockchip-vop ff930000.vop: Failed to attach iommu device
[  277.665268] rockchip-vop ff930000.vop: failed to attach dma mapping, -110
```

I have uploaded a short sample where i see this issue 

https://www.dropbox.com/s/grxtyzyzq8dj6i4/My%20Little%20Pony-%20Friendship%20is%20Magic%20-%20S05E04%20-%20Bloom%20and%20Gloom%20SDTV.mp4?dl=0

LongChair, 2017-06-14:
The patch which was provided below seems t osolve the playback start issue 

```
diff --git a/drivers/dma/pl330.c b/drivers/dma/pl330.c
index 73e0a3f..42cff5d 100644
--- a/drivers/dma/pl330.c
+++ b/drivers/dma/pl330.c
@@ -1356,6 +1356,29 @@ static inline int _loop_cyclic(struct pl330_dmac *pl330, unsigned dry_run,
 		off += _emit_LPEND(dry_run, &buf[off], &lpend);
 	}
 
+	if (pl330->peripherals_req_type == BURST) {
+		unsigned int ccr = pxs->ccr;
+		unsigned long c = 0;
+
+		c = BYTE_MOD_BURST_LEN(x->bytes, pxs->ccr);
+
+		if (c) {
+			ccr &= ~(0xf << CC_SRCBRSTLEN_SHFT);
+			ccr &= ~(0xf << CC_DSTBRSTLEN_SHFT);
+			off += _emit_MOV(dry_run, &buf[off], CCR, ccr);
+			//off += _loop(pl330, dry_run, &buf[off], &c, pxs);
+			off += _emit_LP(dry_run, &buf[off], 1, c - 1);
+			ljmp1 = off;
+			off += _bursts(pl330, dry_run, &buf[off], pxs, 1);
+			lpend.cond = ALWAYS;
+			lpend.forever = false;
+			lpend.loop = 1;
+			lpend.bjump = off - ljmp1;
+			off += _emit_LPEND(dry_run, &buf[off], &lpend);
+			off += _emit_MOV(dry_run, &buf[off], CCR, pxs->ccr);
+		}
+	}
+
 	off += _emit_SEV(dry_run, &buf[off], ev);
 
 	lpend.cond = ALWAYS;
@@ -1457,13 +1480,13 @@ static int _setup_req(struct pl330_dmac *pl330, unsigned dry_run,
 
 	x = &pxs->desc->px;
 
-	if (!pxs->desc->cyclic) {
-		if (pl330->peripherals_req_type != BURST) {
-			/* Error if xfer length is not aligned at burst size */
-			if (x->bytes % (BRST_SIZE(pxs->ccr) * BRST_LEN(pxs->ccr)))
-				return -EINVAL;
-		}
+	if (pl330->peripherals_req_type != BURST) {
+		/* Error if xfer length is not aligned at burst size */
+		if (x->bytes % (BRST_SIZE(pxs->ccr) * BRST_LEN(pxs->ccr)))
+			return -EINVAL;
+	}
 
+	if (!pxs->desc->cyclic) {
 		off += _setup_xfer(pl330, dry_run, &buf[off], pxs);
 
 		/* DMASEV peripheral/event */
@@ -1471,10 +1494,6 @@ static int _setup_req(struct pl330_dmac *pl330, unsigned dry_run,
 		/* DMAEND */
 		off += _emit_END(dry_run, &buf[off]);
 	} else {
-		/* Error if xfer length is not aligned at burst size */
-		if (x->bytes % (BRST_SIZE(pxs->ccr) * BRST_LEN(pxs->ccr)))
-			return -EINVAL;
-
 		off += _setup_xfer_cyclic(pl330, dry_run, &buf[off],
 						pxs, thrd->ev);
 	}
```

[... 558 lines cut ...]
[   47.805635] 0x000003a0: 00000000 00000000 00000000 00000000
[   47.805635] 0x000003b0: 00000000 00000000 00000000 00000000
[   47.805635] 0x000003c0: 00000000 00000000 00000000 00000000
[   47.805635] 0x000003d0: 00000000 00000000 00000000 00000000
[   47.805635] 0x000003e0: 00000000 00000000 00000000 00000000
[   47.805764]     Connector: HDMI-A
[   47.805768] 	bus_format[100a] output_mode[f]
[   47.805773]     Display mode: 4096x2160p30
[   47.805778] 	clk[297000] real_clk[297000] type[40] flag[5]
[   47.805782] 	H: 4096 4184 4272 4400
[   47.805786] 	V: 2160 2168 2178 2250
[   47.805791]     win0-0: DISABLED
[   47.805796]     win1-0: ACTIVE
[   47.805802] 	format: AR24 little-endian (0x34325241)
[   47.805806] 	zpos: 1
[   47.805810] 	src: pos[0x0] rect[1920x1080]
[   47.805815] 	dst: pos[0x0] rect[4096x2160]
[   47.805820] 	buf[0]: addr: 0x00000000029ad000 pitch: 7680 offset: 0
[   47.805825]     win2-0: ACTIVE
[   47.805830] 	format: AR24 little-endian (0x34325241)
[   47.805834] 	zpos: 2
[   47.805838] 	src: pos[0x0] rect[64x64]
[   47.805844] 	dst: pos[0x0] rect[64x64]
[   47.805848] 	buf[0]: addr: 0x00000000021c0000 pitch: 256 offset: 0
```

LongChair, 2017-06-14:
Here is a full kernel log file where this issue happenned :

[kernel.txt](https://github.com/Kwiboo/linux-rockchip/files/1074359/kernel.txt)

yanghanxing, 2017-06-20:
Did you modify dma code yourselves?
@ayaka said he can't meet by use gstreamer.

LongChair, 2017-06-20:
No i don't think we changed anything regarding dma. I know @Kwiboo mentionned that recent changes to pl330 seems to have introduced such issues.

yanghanxing, 2017-07-04:
[0001-arm64-dts-rk3328-enable-the-power-domain-for-decoder.txt](https://github.com/Kwiboo/linux-rockchip/files/1120863/0001-arm64-dts-rk3328-enable-the-power-domain-for-decoder.txt)

yanghanxing, 2017-07-04:
try it

Kwiboo, 2017-07-30:
The power domain change did not have much affect but I have not seen this issue with latest kernel updates. #20 might be related to this issue.

closed 2017-07-30 by Kwiboo