Kwiboo/linux-rockchip #14: RK3328 : device locks on reboot

LongChair, 2017-06-15:
Quite often when rebooting with `reboot`, it seems that the device gets locked.

It seems it won't happen if we boot in console mode only.
The only way to restart the device then is to pull the plug.

I think kwiboo saw it stuck on "Starting kernel", @Kwiboo can you confirm ?

LongChair, 2017-06-15:
Here is a log thta @Kwiboo posted some time ago :

```
LibreELEC:~ # [  307.820028] reboot: Restarting system
INFO:    PSCI Power Domain Map:
INFO:      Domain Node : Level 2, parent_node -1, State ON (0x0)
INFO:      Domain Node : Level 1, parent_node 0, State ON (0x0)
INFO:      Domain Node : Level 0, parent_node 0, State ON (0x0)
INFO:      Domain Node : Level 0, parent_node 0, State ON (0x0)
INFO:      CPU Node : MPID 0x0, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x1, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x2, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x3, parent_node 1, State ON (0x0)
DDR version 1.06 20170424
In
SRX
DDR3
786MHz
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
ddrconfig:6
OUT
Boot1 Release Time: 2017-05-18, version: 2.43
ChipType = 0x11, 186
SdmmcInit=2 0
BootCapSize=2000
UserCapSize=15028MB
FwPartOffset=2000 , 2000
SdmmcInit=0 0
BootCapSize=0
UserCapSize=15193MB
FwPartOffset=2000 , 0
StorageInit ok = 59216
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
LoadTrustBL
No find bl30.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0x62060
RunBL31 0x10000
NOTICE:  BL31: v1.3(release):aa660e1
NOTICE:  BL31: Built : 16:38:31, Mar  9 2017
NOTICE:  BL31: Release version: v1.1
INFO:    ARM GICv2 driver initialized
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
INFO:    BL31: Initializing BL32
INF [0x0] TEE-CORE:init_primary_helper:336: Initializing (1.1.0-68-gf89f3ad-dev #4 Fri Mar  3 00:43:47 UTC 2017 aarch64)
INF [0x0] TEE-CORE:init_primary_helper:337: Release version: 1.1
INF [0x0] TEE-CORE:init_teecore:83: teecore inits done
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9
U-Boot 2017.05 (May 27 2017 - 12:46:11 +0200)
Model: Rockchip RK3328 EVB
DRAM:  2 GiB
MMC:   rksdmmc@ff500000: 1, rksdmmc@ff520000: 0
*** Warning - bad CRC, using default environment
In:    serial@ff130000
Out:   serial@ff130000
Err:   serial@ff130000
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
173 bytes read in 3 ms (55.7 KiB/s)
1:      LibreELEC
Retrieving file: /Image
reading /Image
20028584 bytes read in 1601 ms (11.9 MiB/s)
append: earlyprintk console=ttyFIQ0 console=tty0 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 debugging textmode ssh
Retrieving file: /rk3328-box.dtb
reading /rk3328-box.dtb
39414 bytes read in 6 ms (6.3 MiB/s)
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Device Tree to 000000007df3f000, end 000000007df4b9f5 ... OK
Starting kernel ...
```

omegamoon, 2017-06-15:
kernel logging of a reboot:

```
[   23.692333] reboot: Restarting system
INFO:    PSCI Power Domain Map:
INFO:      Domain Node : Level 2, parent_node -1, State ON (0x0)
INFO:      Domain Node : Level 1, parent_node 0, State ON (0x0)
INFO:      Domain Node : Level 0, parent_node 0, State ON (0x0)
INFO:      Domain Node : Level 0, parent_node 0, State ON (0x0)
INFO:      CPU Node : MPID 0x0, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x1, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x2, parent_node 1, State ON (0x0)
INFO:      CPU Node : MPID 0x3, parent_node 1, State ON (0x0)
DDR version 1.06 20170424
In
SRX
DDR3
933MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
ddrconfig:1
OUT
Boot1 Release Time: 2017-05-18, version: 2.43
ChipType = 0x11, 184
SdmmcInit=2 0
BootCapSize=2000
UserCapSize=7456MB
FwPartOffset=2000 , 2000
SdmmcInit=0 2
StorageInit ok = 35286
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
LoadTrustBL
No find bl30.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0x57bcc
RunBL31 0x10000
NOTICE:  BL31: v1.3(debug):3cb76c2
NOTICE:  BL31: Built : 21:10:34, Feb 23 2017
NOTICE:  BL31: Release version: v1.1
INFO:    ARM GICv2 driver initialized
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.02-RK3328-06-g84e0a7b (Apr 20 2017 - 19:05:18)

CPU: rk3328
cpu version = 1
CPU's clock information:
    arm pll = 600000000HZ
    general pll = 800000000HZ
    ddr pll = 1848000000HZ
    codec pll = 594000000HZ
    new pll = 594000000HZ
Board:	Rockchip platform Board
Uboot as second level loader
DRAM:  Found dram banks: 1
Adding bank:0000000000200000(000000003fe00000)
128 MiB
GIC CPU mask = 0x00000001
SdmmcInit = 0 20
SdmmcInit = 2 0
storage init OK!
Using default environment

GetParam
remotectl v0.1
pwm freq=0x11b3dc
pwm_freq_nstime=0x35d
Load FDT from boot image.
no key node
pmic:rk805
can't find dts node for pwm0
```

yanghanxing, 2017-06-20:
Did you meet this issue on Rock64?
I can't find on my RK3328 EVB board.

LongChair, 2017-06-20:
We met that issue with Z28, so far we didn't recieve Rock64 boards yet.

LongChair, 2017-06-25:
I didn't seem to have those locks anymore with the latest changes, will close it for now, we'll reopen it if we get some proper way to reporduce it.

closed 2017-06-25 by LongChair

yanghanxing, 2017-06-26:
OK，It‘s good news.