Kwiboo/linux-rockchip #15: RK3328 : 4K doesn't get any modes over 30 fps.

LongChair, 2017-06-17:
When I connect my RK3328 box to the 4K display i have which supports 4K@60 modes, i can't seem to get that mode.

The following modes will be listed :
```
cat /sys/class/drm/card0-HDMI-A-1/modes
4096x2160p30
4096x2160p30
4096x2160p25
4096x2160p24
4096x2160p24
3840x2160p30
3840x2160p30
3840x2160p25
3840x2160p24
3840x2160p24
1920x1080p60
1920x1080p60
1920x1080i60
1920x1080i60
1920x1080p50
1920x1080i50
1920x1080i48
1920x1080p30
1920x1080p30
1920x1080p25
1920x1080p24
1920x1080p24
1280x720p60
1280x720p60
1280x720p50
800x600p60
720x576p50
720x576i50
720x480p60
720x480i60
```

I grabbed the edid from :
```
hexdump /sys/class/drm/card0-HDMI-A-1/edid
0000000 ff00 ffff ffff 00ff b358 3700 0000 0000
0000010 1901 0301 5f80 7836 cf0a a374 4c57 23b0
0000020 4809 af4c 80ef 00b3 0095 40a9 4090 0081
0000030 8081 4081 0101 e808 3000 70f2 805a 58b0
0000040 008a 1d50 0074 1e00 3a02 1880 3871 402d
0000050 2c58 0045 1d50 0074 1e00 0000 fc00 3400
0000060 5533 4448 4c5f 4443 545f 0a56 0000 fd00
0000070 3000 0f3e 3c46 0a00 2020 2020 2020 c301
0000080 0302 f147 015c 0706 0302 1615 1211 0413
0000090 0514 901f 2120 5d22 5f5e 6160 6362 6564
00000a0 2c66 0709 1507 5007 063f 57c0 0006 0183
00000b0 0000 036e 000c 0020 3cf8 0020 0480 0203
00000c0 e501 000f 6000 010c 801d 733e 2d38 7e40
00000d0 452c 0080 52d0 0000 011e 801d 72d0 2d1c
00000e0 1020 252c 0080 52d0 0000 009e 0000 0000
00000f0 0000 0000 0000 0000 0000 0000 0000 f800
0000100
```

@kwiboo showed me a site where you can get this interpreted http://www.edidreader.com/ and the EDID seems to report correct modes there, but they are not usable on 3328.

@yanghanxing : any thoughts ?

LongChair, 2017-06-21:
@zheng2012 : let me know if there is anything i can test to fix this :)

Kwiboo, 2017-06-25:
At https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/rockchip/dw_hdmi-rockchip.c#L459-L460 there is a check for 300Mhz that limits the use of 4K 50/60hz modes.

LongChair, 2017-07-18:
@yanghanxing : i tried the patch you provided on ROCK64.

It gives somehow different results as it will not list the 4K@60Hz modes and will tell it is the current mode.
Although my 4K TV will say there is no signal.

I grabbed the kernel log :

[kernel.txt](https://github.com/Kwiboo/linux-rockchip/files/1154851/kernel.txt)

Kwiboo, 2017-07-30:
The recent kernel update seems to detect a lot more modes, but all modes do not seem to work and produce 'no-signal' message on TV, I will raise a new issue regarding non-working modes.

**EDID**
```
LibreELEC:~ # hexdump -C /sys/class/drm/card0-HDMI-A-1/edid
00000000  00 ff ff ff ff ff ff 00  4d d9 03 c8 01 01 01 01  |........M.......|
00000010  01 19 01 03 80 90 51 78  0a 0d c9 a0 57 47 98 27  |......Qx....WG.'|
00000020  12 48 4c 21 08 00 81 80  a9 c0 71 4f b3 00 01 01  |.HL!......qO....|
00000030  01 01 01 01 01 01 02 3a  80 18 71 38 2d 40 58 2c  |.......:..q8-@X,|
00000040  45 00 9f 29 53 00 00 1e  01 1d 00 72 51 d0 1e 20  |E..)S......rQ.. |
00000050  6e 28 55 00 9f 29 53 00  00 1e 00 00 00 fc 00 53  |n(U..)S........S|
00000060  4f 4e 59 20 54 56 20 20  2a 30 32 0a 00 00 00 fd  |ONY TV  *02.....|
00000070  00 30 3e 0e 46 3c 00 0a  20 20 20 20 20 20 01 df  |.0>.F<..      ..|
00000080  02 03 60 f0 5b 61 60 5d  5e 5f 62 1f 10 14 05 13  |..`.[a`]^_b.....|
00000090  04 20 22 3c 3e 12 16 03  07 11 15 02 06 01 65 66  |. "<>.........ef|
000000a0  29 0d 7f 07 15 07 50 3d  07 bc 83 0f 00 00 78 03  |).....P=......x.|
000000b0  0c 00 10 00 b8 3c 2f d0  8a 01 02 03 04 01 40 1f  |.....</.......@.|
000000c0  c0 80 90 d0 e0 f0 d6 67  d8 5d c4 01 78 80 01 e2  |.......g.]..x...|
000000d0  00 f9 e3 05 ff 01 e5 0f  03 00 00 06 e3 06 0d 01  |................|
000000e0  01 1d 80 18 71 1c 16 20  58 2c 25 00 9f 29 53 00  |....q.. X,%..)S.|
000000f0  00 9e 00 00 00 00 00 00  00 00 00 00 00 00 00 62  |...............b|
```

Modes
```
LibreELEC:~ # cat /sys/class/drm/card0-HDMI-A-1/modes
1920x1080p60
4096x2160p60
4096x2160p60
4096x2160p60
4096x2160p50
4096x2160p50
4096x2160p24
4096x2160p24
3840x2160p60
3840x2160p60
3840x2160p60
3840x2160p50
3840x2160p50
3840x2160p30
3840x2160p30
3840x2160p25
3840x2160p24
3840x2160p24
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080p50
1920x1080p50
1920x1080p50
1920x1080i50
1920x1080i50
1920x1080i50
1920x1080i50
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p50
1280x720p50
1280x720p50
1280x720p50
1280x720p30
1280x720p30
1280x720p30
1280x720p30
1280x720p24
1280x720p24
1280x720p24
1280x720p24
800x600p60
720x576p50
720x576i50
720x480p60
720x480i60
```

closed 2017-07-30 by Kwiboo

prisoner881, 2018-08-04:
Anything new on this?  My PINE64 RK3328 shows this after booting LibreELEC Alpha 3 from MicroSD:

LibreELEC:~ # cat /sys/class/drm/card0-HDMI-A-1/modes
1280x720p60
1920x1080p60
1920x1080p50
1280x720p50
720x576p50
720x480p60

This is hooked directly to an LG 4K TV.  My prior ODroid C2 automatically recognized the 4K resolutions but the RK3328 will not.  I don't know if this is a hardware issue or something to do with the LibreELEC Alpha (I suspect the Alpha).

Is there a way to force the 4K resolutions to appear?