Kwiboo/linux-rockchip #16: RK3288/RK3328 : Changes to pl330 cause media playback issues

Kwiboo, 2017-06-19:
Recent changes to pl330 in `release-4.4` causes media playback issues and page faults for iommu/vop.

Reverting Rockchip changes and backporting upstream pl330 from mainline linux seems to solve this issue, see https://github.com/Kwiboo/linux-rockchip/compare/rockchip-4.4...rockchip-4.4-pl330 for the tested revert+backport.

Example log of the issue:
```
[  912.190671] rk_iommu ff930300.iommu: Page fault at 0x019e7400 of type read
[  912.190680] rk_iommu ff930300.iommu: iova = 0x019e7400: dte_index: 0x6 pte_index: 0x1e7 page_offset: 0x400
[  912.190688] rk_iommu ff930300.iommu: mmu_dte_addr: 0x2e1d4000 dte@0x2e1d4018: 0x2693c001 valid: 1 pte@0x2693c79c: 0x1c68b006 valid: 0 page@0x00000000 flags: 0x0
[  912.190694] VOP [ff940000.vop]: DISABLED
[  912.190697]
[  912.190697] 0x00000000: 00000001 03007236 00803000 00000000
[  912.190697] 0x00000010: 000000bf 0000e401 00000000 00711c08
[  912.190697] 0x00000020: 00461020 00000000 00000000 00000000
[  912.190697] 0x00000030: 00000029 45400000 00000000 01e001e0
[  912.190697] 0x00000040: 019bc000 01b3a800 0327077f 0327077f
[  912.190697] 0x00000050: 00b100c0 10001000 7fae7fdd 00000000
[  912.190697] 0x00000060: 00000000 00000000 00000000 00000000
[  912.190697] 0x00000070: 00000081 00400000 00000000 01400780
[  912.190697] 0x00000080: 00fd2000 00000000 0437077f 0437077f
[  912.190697] 0x00000090: 002900c0 10001000 10001000 00000000
[  912.190697] 0x000000a0: 00000069 000000c0 00000000 00000000
[  912.190697] 0x000000b0: 00000001 00000000 01400140 01400140
[  912.190697] 0x000000c0: 00000000 00ef013f 000a000a 00000000
[  912.190697] 0x000000d0: 00000000 00ef013f 000a000a 00000000
00000001 00000000 01400140 01400140
[  912.190826] 0x00000110: 00000000 00ef013f 000a000a 00000000
[  912.190826] 0x00000120: 00000000 00ef013f 000a000a 00000000
[  912.190826] 0x00000130: 00000000 00ef013f 000a000a 00000000
[  912.190826] 0x00000140: 00000000 00ef013f 000a000a 00000000
[  912.190826] 0x00000150: 00000000 00000000 00000000 000a000a
[  912.190826] 0x00000160: 00000000 00000000 00000000 00000000
[  912.190826] 0x00000170: 00c00840 00290461 10001000 00000000
[  912.190826] 0x00000180: 00000003 000a00fa 0abe002c 00c00840
[  912.190826] 0x00000190: 04650005 00290461 00000000 VOP [ff930000.vop]: ACTIVE
[  912.190904]     Connector: HDMI-A
[  912.190907]  bus_format[100a] output_mode[f]
[  912.190911]     Display mode: 1920x1080p24
[  912.190915]  clk[74250] real_clk[74250] type[40] flag[5]
[  912.190919]  H: 1920 2558 2602 2750
[  912.190922]  V: 1080 1084 1089 1125
[  912.190925]     win0-0: DISABLED
[  912.190929]     win1-0: ACTIVE
[  912.190933]  format: AR24 little-endian (0x34325241)
[  912.190936]  zpos: 1
[  912.190939]  src: pos[0x0] rect[1920x1080]
[  912.190943]  dst: pos[0x0] rect[1920x1080]
[  912.190947]  buf[0]: addr: 0x00fd2000 pitch: 7680 offset: 0
[  912.190950]     win2-0: DISABLED
[  912.190953]     win2-0: DISABLED
[  912.190956]     win2-1: DISABLED
[  912.190959]     win2-2: DISABLED
[  912.190962]     win3-0: DISABLED
[  912.190966]     win3-0: DISABLED
[  912.190969]     win3-1: DISABLED
[  912.190972]     win3-2: DISABLED
[  912.190982] rk_iommu ff930300.iommu: Page fault at 0x01b51780 of type read
[  912.190989] rk_iommu ff930300.iommu: iova = 0x01b51780: dte_index: 0x6 pte_index: 0x351 page_offset: 0x780
[  912.190996] rk_iommu ff930300.iommu: mmu_dte_addr: 0x2e1d4000 dte@0x2e1d4018: 0x2693c001 valid: 1 pte@0x2693cd44: 0x1efbd006 valid: 0 page@0x00000000 flags: 0x0
[  912.191000] VOP [ff940000.vop]: DISABLED
[  912.191003]
[  912.191003] 0x00000000: 00000001 03007236 00803000 00000000
[  912.191003] 0x00000010: 000000bf 0000e401 00000000 00711c08
[  912.191003] 0x00000020: 00461020 00000000 00000000 00000000
[  912.191003] 0x00000030: 00000029 45400000 00000000 01e001e0
[  912.191003] 0x00000040: 019bc000 01b3a800 0327077f 0327077f
[  912.191003] 0x00000050: 00b100c0 10001000 7fae7fdd 00000000
[  912.191003] 0x00000060: 00000000 00000000 00000000 00000000
[  912.191003] 0x00000070: 00000081 00400000 00000000 01400780
[  912.191003] 0x00000080: 00fd2000 00000000 0437077f 0437077f
[  912.191003] 0x00000090: 002900c0 10001000 10001000 00000000
[  912.191003] 0x000000a0: 00000069 000000c0 00000000 00000000
[  912.191003] 0x000000b0: 00000001 00000000 01400140 01400140
[  912.191003] 0x000000c0: 00000000 00ef013f 000a000a 00000000
[  912.191003] 0x000000d0: 00000000 00ef013f 000a000a 00000000
00000001 00000000 01400140 01400140
[  912.191124] 0x00000110: 00000000 00ef013f 000a000a 00000000
[  912.191124] 0x00000120: 00000000 00ef013f 000a000a 00000000
[  912.191124] 0x00000130: 00000000 00ef013f 000a000a 00000000
[  912.191124] 0x00000140: 00000000 00ef013f 000a000a 00000000
[  912.191124] 0x00000150: 00000000 00000000 00000000 000a000a
[  912.191124] 0x00000160: 00000000 00000000 00000000 00000000
[  912.191124] 0x00000170: 00c00840 00290461 10001000 00000000
[  912.191124] 0x00000180: 00000003 000a00fa 0abe002c 00c00840
[  912.191124] 0x00000190: 04650005 00290461 00000000 VOP [ff930000.vop]: ACTIVE
[  912.191198]     Connector: HDMI-A
[  912.191201]  bus_format[100a] output_mode[f]
[  912.191205]     Display mode: 1920x1080p24
[  912.191209]  clk[74250] real_clk[74250] type[40] flag[5]
[  912.191212]  H: 1920 2558 2602 2750
[  912.191215]  V: 1080 1084 1089 1125
[  912.191218]     win0-0: DISABLED
[  912.191221]     win1-0: ACTIVE
[  912.191226]  format: AR24 little-endian (0x34325241)
[  912.191228]  zpos: 1
[  912.191231]  src: pos[0x0] rect[1920x1080]
[  912.191235]  dst: pos[0x0] rect[1920x1080]
[  912.191238]  buf[0]: addr: 0x00fd2000 pitch: 7680 offset: 0
[  912.191242]     win2-0: DISABLED
[  912.191245]     win2-0: DISABLED
[  912.191247]     win2-1: DISABLED
[  912.191254]     win2-2: DISABLED
[  912.191257]     win3-0: DISABLED
[  912.191260]     win3-0: DISABLED
[  912.191263]     win3-1: DISABLED
[  912.191266]     win3-2: DISABLED
[  912.202080] drm_crtc_check_viewport: skip check
[  912.231201] rk_iommu ff930300.iommu: Enable stall request timed out, status: 0x00004b
[  912.241196] rk_iommu ff930300.iommu: Disable paging request timed out, status: 0x00004b
[  912.304581] rk_iommu ff930300.iommu: Enable stall request timed out, status: 0x00004b
```

yanghanxing, 2017-06-20:
Have you modified dma code by yourselves?

Kwiboo, 2017-06-20:
We should not have any modified dma code apart from testing this revert code to use a `pl330.c` version very close to mainline linux.

If I use the unmodified `pl330.c` from tip of `release-4.4` I usually see the iommu page fault and VOP issue posted above.

LongChair, 2017-06-21:
I checked removing the pl330 revert patch this morning, and i don't seem to get any crash issue with a few videos i tested. @Kwiboo did you get a chance to try it again ?

LongChair, 2017-06-21:
scratch that, i was able to reproduce it. If i connect my box to 1080P AVR -> 4K TV i don't get it.
If i connect the box to the 4K TV i get it all the time upon playback stop with the following log 
[kernel.txt](https://github.com/Kwiboo/linux-rockchip/files/1090338/kernel.txt)

LongChair, 2017-06-21:
@kwiboo : also reappying the patch https://github.com/Kwiboo/LibreELEC.tv/blob/rockchip/projects/Rockchip/patches/linux/rockchip-4.4/linux-1000-pl330.patch doesn't seem to fix this issue when stopping playback.

But i'm only experiencing this in 4K and not in 1080p.

LongChair, 2017-06-21:
after a few more test, i was also able to reproduce that in 1080p and with kwiboo pl330 revert patch, so that issue is still there

LongChair, 2017-06-21:
I think I have a good way to reproduce it.
 - From cold boot in 1080p, start a playabck several times : No issue
 - Unplug HDMI cable while the device is powered and replug it again 
 - Play the same file again, it starts fine, but when you stop it, it will crash.

@yanghanxing : could you try that scenario and let me know if you can reproduce that issue ?

yanghanxing, 2017-07-04:
[0001-arm64-dts-rk3328-enable-the-power-domain-for-decoder.txt](https://github.com/Kwiboo/linux-rockchip/files/1120861/0001-arm64-dts-rk3328-enable-the-power-domain-for-decoder.txt)

yanghanxing, 2017-07-04:
pls use this patch to test

Kwiboo, 2017-07-30:
The power domain patch did not made any difference but something in recent kernel update have made the  iommu/vop page fault go away.

There is still some issues with seeking, buffering and audio rate unless we patch pl330 but will raise another issue to track that issue.

closed 2017-07-30 by Kwiboo