Kwiboo/linux-rockchip #18: RK3328 : device locks up after a while

omegamoon, 2017-06-23:
The rk3328 device currently locks up after a while. sometimes it reboots, and other times it just locks up.

I collected a couple of stack traces while playing media in Kodi which I will add separately. When the device is idle, doing nothing, it also locks up after a while, but then I see no stack trace at all... it just locks up.

Currently it looks as if the device locks up after approximately 15-20mins... always!

LongChair, 2017-06-25:
with the latest changes I had rock64 to run a lot longer. 
@omegamoon : could you check if you still have that behavior with those ?

Kwiboo, 2017-06-25:
the ~15-20 min lockup issue have also been resolved on my my Z28 with the latest changes, I think we can close this once @omegamoon have confirmed the same

yanghanxing, 2017-06-26:
The trust.img update should resolve the issue.

omegamoon, 2017-06-28:
no lockups seen after applying last changes.
closing this issue

closed 2017-06-28 by omegamoon

suppenkasper0815, 2017-11-23:
@omegamoon Could you please send me some hints for Alfawise Z28 Pro (RK3228) and RK3229-Images per PM?