Kwiboo/linux-rockchip #19: RK3328 : Not all HDMI modes works

Kwiboo, 2017-07-30:
With the changes in latest kernel update the ROCK64 device started to report 4K60hz modes, but not all modes reported seems to work.

For 1080p only 60hz works on my Sony KD-65X8509C TV.
50, 30, 25 and 24 hz do not produce any picture.

All modes (including above) reported on my Tinker Board running same kernel work using same HDMI-port and cable.

**EDID**
```
LibreELEC:~ # hexdump -C /sys/class/drm/card0-HDMI-A-1/edid
00000000  00 ff ff ff ff ff ff 00  4d d9 03 c8 01 01 01 01  |........M.......|
00000010  01 19 01 03 80 90 51 78  0a 0d c9 a0 57 47 98 27  |......Qx....WG.'|
00000020  12 48 4c 21 08 00 81 80  a9 c0 71 4f b3 00 01 01  |.HL!......qO....|
00000030  01 01 01 01 01 01 02 3a  80 18 71 38 2d 40 58 2c  |.......:..q8-@X,|
00000040  45 00 9f 29 53 00 00 1e  01 1d 00 72 51 d0 1e 20  |E..)S......rQ.. |
00000050  6e 28 55 00 9f 29 53 00  00 1e 00 00 00 fc 00 53  |n(U..)S........S|
00000060  4f 4e 59 20 54 56 20 20  2a 30 32 0a 00 00 00 fd  |ONY TV  *02.....|
00000070  00 30 3e 0e 46 3c 00 0a  20 20 20 20 20 20 01 df  |.0>.F<..      ..|
00000080  02 03 60 f0 5b 61 60 5d  5e 5f 62 1f 10 14 05 13  |..`.[a`]^_b.....|
00000090  04 20 22 3c 3e 12 16 03  07 11 15 02 06 01 65 66  |. "<>.........ef|
000000a0  29 0d 7f 07 15 07 50 3d  07 bc 83 0f 00 00 78 03  |).....P=......x.|
000000b0  0c 00 10 00 b8 3c 2f d0  8a 01 02 03 04 01 40 1f  |.....</.......@.|
000000c0  c0 80 90 d0 e0 f0 d6 67  d8 5d c4 01 78 80 01 e2  |.......g.]..x...|
000000d0  00 f9 e3 05 ff 01 e5 0f  03 00 00 06 e3 06 0d 01  |................|
000000e0  01 1d 80 18 71 1c 16 20  58 2c 25 00 9f 29 53 00  |....q.. X,%..)S.|
000000f0  00 9e 00 00 00 00 00 00  00 00 00 00 00 00 00 62  |...............b|
```

**Modes (include 4:2:0 and fractal clock modes)**
```
LibreELEC:~ # cat /sys/class/drm/card0-HDMI-A-1/modes
1920x1080p60
4096x2160p60
4096x2160p60
4096x2160p60
4096x2160p50
4096x2160p50
4096x2160p24
4096x2160p24
3840x2160p60
3840x2160p60
3840x2160p60
3840x2160p50
3840x2160p50
3840x2160p30
3840x2160p30
3840x2160p25
3840x2160p24
3840x2160p24
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080p60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080i60
1920x1080p50
1920x1080p50
1920x1080p50
1920x1080i50
1920x1080i50
1920x1080i50
1920x1080i50
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p30
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1920x1080p24
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p60
1280x720p50
1280x720p50
1280x720p50
1280x720p50
1280x720p30
1280x720p30
1280x720p30
1280x720p30
1280x720p24
1280x720p24
1280x720p24
1280x720p24
800x600p60
720x576p50
720x576i50
720x480p60
720x480i60
```

yanghanxing, 2017-07-31:
I will check it.
Please check the /sys/class/display/HDMI/debug node info first, and provide it to me.

yanghanxing, 2017-08-01:
check this node : /sys/class/drm/card0-HDMI-A-1/debug

yanghanxing, 2017-08-01:
Please use this patch for the issue:
`From 857041ee2b1814c4dc3962b69344c67c4da79c0a Mon Sep 17 00:00:00 2001
From: xuhuicong <xhc@rock-chips.com>
Date: Fri, 23 Jun 2017 18:56:17 +0800
Subject: [PATCH] drm/rockchip: hdmi: fix no sound some time

Change-Id: Ic9f931d9a5b7bca954363293a20ca242eb0bfa6f

---

diff --git a/drivers/gpu/drm/bridge/dw-hdmi.c b/drivers/gpu/drm/bridge/dw-hdmi.c
index 168c276..a8a2edf 100644
--- a/drivers/gpu/drm/bridge/dw-hdmi.c
+++ b/drivers/gpu/drm/bridge/dw-hdmi.c
@@ -1800,10 +1800,6 @@
 		HDMI_FC_INVIDCONF_IN_I_P_INTERLACED :
 		HDMI_FC_INVIDCONF_IN_I_P_PROGRESSIVE;
 
-	inv_val |= hdmi->sink_is_hdmi ?
-		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
-		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
-
 	hdmi_writeb(hdmi, inv_val, HDMI_FC_INVIDCONF);
 
 	hdisplay = mode->hdisplay;
@@ -2075,6 +2071,9 @@
 	/* not for DVI mode */
 	if (hdmi->sink_is_hdmi) {
 		dev_dbg(hdmi->dev, "%s HDMI mode\n", __func__);
+		hdmi_modb(hdmi, HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF);
 
 		/* HDMI Initialization Step F - Configure AVI InfoFrame */
 		hdmi_config_AVI(hdmi, mode);
`

yanghanxing, 2017-08-01:
> From 857041ee2b1814c4dc3962b69344c67c4da79c0a Mon Sep 17 00:00:00 2001
From: xuhuicong <xhc@rock-chips.com>
Date: Fri, 23 Jun 2017 18:56:17 +0800
Subject: [PATCH] drm/rockchip: hdmi: fix no sound some time

Change-Id: Ic9f931d9a5b7bca954363293a20ca242eb0bfa6f

---

diff --git a/drivers/gpu/drm/bridge/dw-hdmi.c b/drivers/gpu/drm/bridge/dw-hdmi.c
index 168c276..a8a2edf 100644
--- a/drivers/gpu/drm/bridge/dw-hdmi.c
+++ b/drivers/gpu/drm/bridge/dw-hdmi.c
@@ -1800,10 +1800,6 @@
 		HDMI_FC_INVIDCONF_IN_I_P_INTERLACED :
 		HDMI_FC_INVIDCONF_IN_I_P_PROGRESSIVE;
 
-	inv_val |= hdmi->sink_is_hdmi ?
-		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
-		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
-
 	hdmi_writeb(hdmi, inv_val, HDMI_FC_INVIDCONF);
 
 	hdisplay = mode->hdisplay;
@@ -2075,6 +2071,9 @@
 	/* not for DVI mode */
 	if (hdmi->sink_is_hdmi) {
 		dev_dbg(hdmi->dev, "%s HDMI mode\n", __func__);
+		hdmi_modb(hdmi, HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF);
 
 		/* HDMI Initialization Step F - Configure AVI InfoFrame */
 		hdmi_config_AVI(hdmi, mode);

yanghanxing, 2017-08-01:
_From 857041ee2b1814c4dc3962b69344c67c4da79c0a Mon Sep 17 00:00:00 2001
From: xuhuicong <xhc@rock-chips.com>
Date: Fri, 23 Jun 2017 18:56:17 +0800
Subject: [PATCH] drm/rockchip: hdmi: fix no sound some time

Change-Id: Ic9f931d9a5b7bca954363293a20ca242eb0bfa6f

---

diff --git a/drivers/gpu/drm/bridge/dw-hdmi.c b/drivers/gpu/drm/bridge/dw-hdmi.c
index 168c276..a8a2edf 100644
--- a/drivers/gpu/drm/bridge/dw-hdmi.c
+++ b/drivers/gpu/drm/bridge/dw-hdmi.c
@@ -1800,10 +1800,6 @@
 		HDMI_FC_INVIDCONF_IN_I_P_INTERLACED :
 		HDMI_FC_INVIDCONF_IN_I_P_PROGRESSIVE;
 
-	inv_val |= hdmi->sink_is_hdmi ?
-		HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE :
-		HDMI_FC_INVIDCONF_DVI_MODEZ_DVI_MODE;
-
 	hdmi_writeb(hdmi, inv_val, HDMI_FC_INVIDCONF);
 
 	hdisplay = mode->hdisplay;
@@ -2075,6 +2071,9 @@
 	/* not for DVI mode */
 	if (hdmi->sink_is_hdmi) {
 		dev_dbg(hdmi->dev, "%s HDMI mode\n", __func__);
+		hdmi_modb(hdmi, HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF_DVI_MODEZ_HDMI_MODE,
+			  HDMI_FC_INVIDCONF);
 
 		/* HDMI Initialization Step F - Configure AVI InfoFrame */
 		hdmi_config_AVI(hdmi, mode);
_

yanghanxing, 2017-08-01:
[857041e.diff.txt](https://github.com/Kwiboo/linux-rockchip/files/1189412/857041e.diff.txt)
Please use this patch to test again

Kwiboo, 2017-08-01:
@yanghanxing thanks!, that patch solved my issue with 1080p modes, but there is still an issue with the 4k 50/60hz modes not getting signal. Will update this issue once I have had time to test more.

yanghanxing, 2017-08-04:
LibreELEC:~ # cat /sys/class/drm/card0-HDMI-A-1/modes
1920x1080p60
4096x2160p60
4096x2160p60
4096x2160p60
4096x2160p50
I see the info you provided, the 4k 50/60hz is supported
Can you give more information about the new issue?

LongChair, 2017-08-27:
I can confirm what @Kwiboo mentioned above, Any mode starting at 3840x2160 with frequencies starting at 50Hz don't display anything (No Signal).
tested it with 3840x2160@60 and 4096x2160@60 ->No signal
Both 3840x2160@30 and 4096x2160@30 will work fine.

Koloss78, 2017-10-08:
In Android i can activate 3840x2160@60 and 4096x2160@60 but not in libreelec

Kwiboo, 2017-10-08:
@Koloss78 please note that this is not a LibreELEC issue tracker and as noted on https://github.com/Kwiboo/LibreELEC.tv/tree/rockchip/projects/Rockchip/devices/ROCK64 supported 4K modes have been limited to 24/30hz, mainly because of the kernel issue reported here.

closed 2020-01-01 by Kwiboo