Kwiboo/linux-rockchip #20: RK3288/RK3328 : DMA pl330 code cause audio and seeking issues

Kwiboo, 2017-07-30:
Ever since https://github.com/rockchip-linux/kernel/commit/e8a6e5086cb82d59cae6ae029b1eb4432cc62288 and related commits there have been issues with audio not working correctly,
Audio will play too fast or too slow, it sounds similar to playing 44khz audio at 22/88khz.

There is also a second issue related to the DMA code affecting seeking while decoding and playing a video.
Sometimes the decoder will get stuck after seeking and won't return any more frames, other times the video wont start playing and Kodi gets stuck buffering until playback is manually stopped.

When using https://github.com/Kwiboo/linux-rockchip/compare/rockchip-4.4...rockchip-4.4-pl330 to revert the changes in `pl330.c` to mainline 4.12 state most of the issues described above gets resolved, the changes to `iommu` did not seem relevant when tested with or without.

Kwiboo, 2017-08-21:
This is still an issue with latest `release-4.4` (https://github.com/rockchip-linux/kernel/commit/03d6a385ab6c876d5cf54261401799c4763360ba) on rk3288/rk3328.

The issue is possibly limited to rk3288/rk3328 (the only SoC I have), as @omegamoon indicated that the `release-4.4` dma code works on rk3399.

LongChair, 2017-08-26:
still and issue with https://github.com/rockchip-linux/kernel/commit/a17856c5e73a4a9113330b089968ba6341bd7870. Waiting in Sugar availability to have a look at this :)

closed 2020-01-01 by Kwiboo