Kwiboo/linux-rockchip #22: RK3328 : HDR media playback issues.

LongChair, 2017-08-10:
We are having issues to play our HDR samples on RK3328.

One of the samples that shows that issue is http://demo-uhd3d.com/fiche.php?cat=uhd&id=159

From what we see, MPP seems to error while parsing the data, this is the log we have been able to retrieve :

```
[ffmpeg] hevc_rkmpp: Initializing RKMPP decoder.
mpi: mpp version:
[ffmpeg] hevc_rkmpp: RKMPP decoder initialized successfully.
[ffmpeg] hevc_rkmpp: Flush.
[ffmpeg] hevc_rkmpp: Wrote 1874 bytes to decoder
H265D_PARSER: No start code is found.
[ffmpeg] hevc_rkmpp: Wrote 705617 bytes to decoder
H265D_PARSER: nal_unit_type: 32, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 33, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 34, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 32, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 33, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 34, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 19, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: nal_unit_type: 32, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 32,len = 24
H265PARSER_PS: Decoding VPS
H265PARSER_PS: vps_id = 0x0
H265PARSER_PS: Main 10 profile bitstream
H265D_PARSER: nal_unit_type: 33, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 33,len = 46
H265PARSER_PS: Decoding SPS
H265PARSER_PS: Main 10 profile bitstream
H265PARSER_PS: read bit left 161
H265PARSER_PS: 2 read bit left 148
H265PARSER_PS: sps->log2_min_cb_size 3
H265PARSER_PS: sps->log2_diff_max_min_coding_block_size 3
H265PARSER_PS: sps->log2_min_tb_size 2
H265PARSER_PS: sps->log2_diff_max_min_transform_block_size 3
H265PARSER_PS: sps->log2_max_trafo_size 5
H265PARSER_PS: sps->amp_enabled_flag = 0
H265PARSER_PS: sps->sao_enabled = 1
H265PARSER_PS: sps->pcm_enabled_flag = 0
H265PARSER_PS: Decoding VUI
H265PARSER_PS: sps->log2_min_cb_size = 3 sps->log2_diff_max_min_coding_block_size = 3
H265PARSER_PS: plus sps->log2_ctb_size 6
H265PARSER_PS: sps->log2_ctb_size 6
H265D_PARSER: nal_unit_type: 34, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 34,len = 7
H265PARSER_PS: Decoding PPS
H265PARSER_PS: Decoding PPS 1
H265PARSER_PS: num bit left 3
H265PARSER_PS: num bit left 2
H265PARSER_PS: num bit left 1
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 39,len = 10
h265d_sei: Decoding SEI
h265d_sei: s->nal_unit_type 39 payload_type 144 payload_size 4
h265d_sei: Skipped PREFIX SEI 144
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 39,len = 30
h265d_sei: Decoding SEI
h265d_sei: s->nal_unit_type 39 payload_type 137 payload_size 24
h265d_sei: mastering_display_colour_volume in
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 39,len = 1754
h265d_sei: Decoding SEI
h265d_sei: s->nal_unit_type 39 payload_type 5 payload_size 1743
h265d_sei: Skipped PREFIX SEI 5
H265D_PARSER: nal_unit_type: 32, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 32,len = 24
H265PARSER_PS: Decoding VPS
H265PARSER_PS: vps_id = 0x0
H265PARSER_PS: Main 10 profile bitstream
H265D_PARSER: nal_unit_type: 33, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 33,len = 46
H265PARSER_PS: Decoding SPS
H265PARSER_PS: Main 10 profile bitstream
H265PARSER_PS: read bit left 161
H265PARSER_PS: 2 read bit left 148
H265PARSER_PS: sps->log2_min_cb_size 3
H265PARSER_PS: sps->log2_diff_max_min_coding_block_size 3
H265PARSER_PS: sps->log2_min_tb_size 2
H265PARSER_PS: sps->log2_diff_max_min_transform_block_size 3
H265PARSER_PS: sps->log2_max_trafo_size 5
H265PARSER_PS: sps->amp_enabled_flag = 0
H265PARSER_PS: sps->sao_enabled = 1
H265PARSER_PS: sps->pcm_enabled_flag = 0
H265PARSER_PS: Decoding VUI
H265PARSER_PS: sps->log2_min_cb_size = 3 sps->log2_diff_max_min_coding_block_size = 3
H265PARSER_PS: plus sps->log2_ctb_size 6
H265PARSER_PS: sps->log2_ctb_size 6
H265D_PARSER: nal_unit_type: 34, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 34,len = 7
H265PARSER_PS: Decoding PPS
H265PARSER_PS: Decoding PPS 1
H265PARSER_PS: num bit left 7
H265PARSER_PS: num bit left 6
H265PARSER_PS: num bit left 5
H265D_PARSER: nal_unit_type: 39, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 39,len = 1754
h265d_sei: Decoding SEI
h265d_sei: s->nal_unit_type 39 payload_type 5 payload_size 1743
h265d_sei: Skipped PREFIX SEI 5
H265D_PARSER: nal_unit_type: 19, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 19,len = 701871
H265D_PARSER: hls_slice_header in
H265D_PARSER: hls_slice_header out
H265_PARSER_REF: width = 3840 height = 2160
H265_PARSER_REF: w_stride 3840 h_stride 2160
H265_PARSER_REF: alloc frame poc 0 slot_index 0
mpp_buf_slot: new width 3840 height 2160 stride hor 4864 ver 2160 fmt    1
H265D_PARSER: decode poc = 0
H265D_PARSER: Decoded frame with POC 0.
H265D_PARSER: Output frame with POC 0 frame->slot_index = 0
[ffmpeg] hevc_rkmpp: Wrote 61183 bytes to decoder
[ffmpeg] hevc_rkmpp: Wrote 2284 bytes to decoder
[ffmpeg] hevc_rkmpp: Wrote 1312 bytes to decoder
[ffmpeg] hevc_rkmpp: Wrote 1493 bytes to decoder
[ffmpeg] hevc_rkmpp: Decoder noticed an info change (3840x2160), format=1
H265D_PARSER: nal_unit_type: 1, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: IS_IRAP frame found error
H265D_PARSER: nal_unit_type: 1, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 1,len = 61179
H265D_PARSER: hls_slice_header in
[ffmpeg] hevc_rkmpp: Received a discard/errinfo frame.
H265D_PARSER: Invalid collocated_ref_idx: 7.
H265D_PARSER: hls_slice_header out
H265D_PARSER: hls_slice_header error ret = -1004
H265D_PARSER: Error parsing NAL unit #0,error ret = 0xd.
H265D_PARSER: current stream is no right skip it
H265D_PARSER: nal_unit_type: 1, nuh_layer_id: 0 temporal_id: 0
[ffmpeg] hevc_rkmpp: Wrote 91034 bytes to decoder
H265D_PARSER: nal_unit_type: 1, nuh_layer_id: 0 temporal_id: 0
H265D_PARSER: s->nal_unit_type = 1,len = 2280
H265D_PARSER: hls_slice_header in
H265D_PARSER: hls_slice_header out
H265D_PARSER: decode poc = 2
H265D_PARSER: nal_unit_type: 0, nuh_layer_id: 0 temporal_id: 0
[ffmpeg] hevc_rkmpp: Wrote 2520 bytes to decoder
```

Kwiboo, 2017-08-23:
Using the following patch seems to make the sample linked to in original post work a little bit better. It also fixes playback of my `Astra-SES_Demo_UHD_satellite_end-of-2015.mkv` sample.
I have no clue as to why this solved the problem or what this hw reg controls.

```
diff --git a/mpp/hal/rkdec/h265d/hal_h265d_reg.c b/mpp/hal/rkdec/h265d/hal_h265d_reg.c
--- a/mpp/hal/rkdec/h265d/hal_h265d_reg.c
+++ b/mpp/hal/rkdec/h265d/hal_h265d_reg.c
@@ -1468,8 +1468,6 @@ MPP_RET hal_h265d_gen_regs(void *hal,  HalTaskInfo *syn)
     }
     hw_regs->sw_interrupt.sw_dec_e         = 1;
     hw_regs->sw_interrupt.sw_dec_timeout_e = 1;
-    hw_regs->sw_interrupt.sw_wr_ddr_align_en = dxva_cxt->pp.tiles_enabled_flag
-                                               ? 0 : 1;

     ///find s->rps_model[i] position, and set register
```

There might also be an issue with the `sw_interrupt`, when comparing the hw regs for h264d/h265d/vp9d `sw_dec_timeout_sta` is missing in `hal_h265d_reg.h`
Below is my best guess at a fix, but without any hw regs documentation it is only a guess.

```
diff --git a/mpp/hal/rkdec/h265d/hal_h265d_reg.h b/mpp/hal/rkdec/h265d/hal_h265d_reg.h
index 1bccb02..432b8db 100644
--- a/mpp/hal/rkdec/h265d/hal_h265d_reg.h
+++ b/mpp/hal/rkdec/h265d/hal_h265d_reg.h
@@ -50,7 +50,8 @@ typedef struct {
     struct swreg_int {
         RK_U32    sw_dec_e            : 1  ;
         RK_U32    sw_dec_clkgate_e    : 1  ;
-        RK_U32    reserve0            : 2  ;
+        RK_U32    reserve0            : 1  ;
+        RK_U32    sw_timeout_mode     : 1  ;
         RK_U32    sw_dec_irq_dis      : 1  ;
         RK_U32    sw_dec_timeout_e    : 1  ;
         RK_U32    sw_buf_empty_en     : 1  ;
@@ -61,8 +62,9 @@ typedef struct {
         RK_U32    sw_dec_rdy_sta      : 1  ;
         RK_U32    sw_dec_bus_sta      : 1  ;
         RK_U32    sw_dec_error_sta    : 1  ;
+        RK_U32    sw_dec_timeout_sta  : 1  ;
         RK_U32    sw_dec_empty_sta    : 1  ;
-        RK_U32    reserve4            : 4  ;
+        RK_U32    reserve3            : 3  ;
         RK_U32    sw_softrst_en_p     : 1  ;
         RK_U32    sw_force_softreset_valid: 1 ;
         RK_U32    sw_softreset_rdy    : 1  ;
```

LongChair, 2017-08-26:
Looks to fix HDR playback issues for me too

yanghanxing, 2017-08-28:
Yeah, Our engineer said it would not impact to the HDR video playback. And we will check it again.

Koloss78, 2017-10-08:
I can start HDR Video but my TV did not Show HDR Content on Display.

Samung TV say UHD 3180x2160 but not HDR UHD 3180x2160.

LongChair, 2018-01-11:
@yanghanxing  : ok here is some update on our HDR investigations and understanding

The following DRM Atomic properties need to be set :

 * `HDR_SOURCE_METADATA` : this is a blob property that needs to be set to the following contents :

```
struct hdr_static_metadata {
    uint16_t eotf;
    uint16_t type;
    uint16_t display_primaries_x[3];
    uint16_t display_primaries_y[3];
    uint16_t white_point_x;
    uint16_t white_point_y;
    uint16_t max_mastering_display_luminance;
    uint16_t min_mastering_display_luminance;
    uint16_t max_fall;
    uint16_t max_cll;
    uint16_t min_cll;
};
```

Most of this information is coming from the media itself.
The first eotf property should be set to the following enumeration :

```
TRADITIONAL_GAMMA_SDR = 0,
TRADITIONA_GAMMA_HDR,
SMPTE_ST2084,
FUTURE_EOTF
```

We can probably feed this information from the stream metadata information.

* `hdmi_output_depth`

We have no info on this one, what is it supposed to be set to ? 

* `hdmi_output_format` 

We have no info on this one, what is it supposed to be set to ? 

* `hdmi_output_colorimetry`

We have no info on this one, what is it supposed to be set to ? 

* `COLOR_SPACE`

We have no info on this one, what is it supposed to be set to ? 

Could you please give us some information regarding those properties like a type and possible values ? 
This would help us moving on :)

Thanks !

yzheng2012, 2018-01-12:
@LongChair ，Hdr_source_metadata is the only property required, other properties are optional. The type field should be zero, and eotf filed should be SMPTE_ST2084.

LongChair, 2018-01-12:
@yzheng2012 : what will happen if i set HDR_SOURCE_METADATA with an HDR movie and that the tv doesn't support it, will it fallback to a supported mode smartly and display it the best it can or will the display be broken ?

mcerveny, 2018-01-25:
Beaware. **Panel "COLOR_SPACE" must be set too.** If "COLOR_SPACE" is leaved in default VOP makes conversions. 
Video output is switched to HDR/BT2020 (tv also say it). The connector read/checked "HDR_PANEL_METADATA" to support EOTF and set "HDR_SOURCE_METADATA" (filled "struct hdr_static_metadata").
```
# cat /sys/kernel/debug/dw-hdmi/status 
PHY: enabled			Mode: HDMI
Pixel Clk: 297000000Hz		TMDS Clk: 297000000Hz
Color Format: YUV422		Color Depth: 10 bit
Colorimetry: ITU.BT2020		EOTF: ST2084
x0: 35400				y0: 14599
x1: 8500				y1: 39850
x2: 6550				y2: 2300
white x: 15635			white y: 16451
max lum: 0			min lum: 0
max cll: 0			max fall: 0
```

- variant "A" with less colors due to y2r-> r2r-> csc[CSC_BT601L]-> post_r2y-> post_sdr2hdr-> post_csc[CSC_BT2020] ([check scheme](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/rockchip/rockchip_drm_vop.c#L711) and [CSC](https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/gpu/drm/rockchip/rockchip_drm_vop.c#L853)).
Win0 plane with NA12 data, set "EOTF"=2 (SMPTE_ST2084) "COLOR_SPACE"=0 (V4L2_COLORSPACE_DEFAULT)
```
# cat /sys/kernel/debug/dri/0/summary 
VOP [ff370000.vop]: ACTIVE
    Connector: HDMI-A
	overlay_mode[0] bus_format[2016] output_mode[f] color_space[10]
    Display mode: 3840x2160p30
	clk[297000] real_clk[297000] type[40] flag[5]
	H: 3840 4016 4104 4400
	V: 2160 2168 2178 2250
    win0-0: ACTIVE
	format: NA12 little-endian (0x3231414e) HDR[2] color_space[0]
	csc: y2r[1] r2r[1] r2y[0] csc mode[1]
	zpos: 0
	src: pos[0x0] rect[3840x2160]
	dst: pos[0x0] rect[3840x2160]
	buf[0]: addr: 0x0000000006b74000 pitch: 4864 offset: 0
	buf[1]: addr: 0x0000000006b74000 pitch: 4864 offset: 10506240
    win1-0: DISABLED
    win2-0: DISABLED
    post: sdr2hdr[1] hdr2sdr[0]
    pre : sdr2hdr[0]
    post CSC: r2y[1] y2r[0] CSC mode[3]
```

- variant "B" with corrected "passthrough" color. Win0 plane is the same except  "COLOR_SPACE"=10 (V4L2_COLORSPACE_BT2020).
```
# cat /sys/kernel/debug/dri/0/summary 
VOP [ff370000.vop]: ACTIVE
    Connector: HDMI-A
	overlay_mode[1] bus_format[2016] output_mode[f] color_space[10]
    Display mode: 3840x2160p30
	clk[297000] real_clk[297000] type[40] flag[5]
	H: 3840 4016 4104 4400
	V: 2160 2168 2178 2250
    win0-0: ACTIVE
	format: NA12 little-endian (0x3231414e) HDR[2] color_space[10]
	csc: y2r[0] r2r[0] r2y[0] csc mode[3]
	zpos: 0
	src: pos[0x0] rect[3840x2160]
	dst: pos[0x0] rect[3840x2160]
	buf[0]: addr: 0x000000000e8b5000 pitch: 4864 offset: 0
	buf[1]: addr: 0x000000000e8b5000 pitch: 4864 offset: 10506240
    win1-0: DISABLED
    win2-0: DISABLED
    post: sdr2hdr[0] hdr2sdr[0]
    pre : sdr2hdr[0]
    post CSC: r2y[0] y2r[0] CSC mode[3]
```

mcerveny, 2018-01-25:
Beware. **You must always set "hdmi_output_format" on connector before sending video** (even with the same value like DRM_HDMI_OUTPUT_YCBCR_LQ). If not re-set "post CSC" chain remains in previous state (eg. when play hdr and then sdr, sdr video becomes magenta colored).

mcerveny, 2018-01-25:
Performance is not perfect. I am still slightly under 60FPS for HDR with optimization (see rockchip-linux/kernel#59).

Kwiboo, 2018-01-25:
@mcerveny thanks for sharing your findings, I have also had some stability issues and seen some artifacts in decoding using the latest kernel and ddr dvfs enabled. I will probably stay on kernel commit https://github.com/rockchip-linux/kernel/commit/9b04710a121d09b7779ba2168e1f07bfbf99f457 that seems more stable in my tests, at least until I have more time to test a more recent kernel commit.

mcerveny, 2018-01-26:
There are problems with HDR/HLG (Hybrid Log-Gamma) EOTF (Electro-Optical Transfer Function)
- at mpp side rockchip-linux/mpp#38 (issue was immediately closed without any solution)
- at drm side rockchip-linux/kernel#61

mcerveny, 2018-01-26:
My test environment:
- testing [kernel](https://github.com/mcerveny/rockchip-linux/tree/release-4.4-hdr)
- testing [utility](https://github.com/mcerveny/utils)

LongChair, 2018-01-26:
@yzheng2012 : could you please confirm Martin’s findings so that we have a clear view on what is required and what’s not ?

mcerveny, 2018-01-27:
HDR/HLG working. Added patch to kernel (see rockchip-linux/kernel#61). MPP patch will not be available (see rockchip-linux/mpp#38). I patched test utility to do workaround (see mcerveny/utils@12b893a7ab051cfb345370774020c3e11dd919a1). Tested (HDR/HLG recognized by LG OLED TV output limited "video=HDMI-A-1:3840x2160@30" due to lack of HDMI2 cables) and running @67FPS without boost CPU/DDR:
```
# ffmpeg -i "LG Cymatic Jazz 4K Demo.ts" -c:v copy -an -bsf:v hevc_mp4toannexb 2k60hdr_hlg.hevc
# ./mpi_dec ../4k60hdr_hlg.hevc 16777220 | grep 'frame(3000)'
FRAME time 222.664861 render_time  11250 us fps=(time(44566471 us)/frame(3000)) 67.32

# echo ---; cat /sys/kernel/debug/dw-hdmi/status; echo ----; cat /sys/kernel/debug/dri/0/summary; echo -----; egrep 'armclk |aclk_rkvdec |cabac |vdec_core |ddrc ' /sys/kernel/debug/clk/clk_summary
---
PHY: enabled			Mode: HDMI
Pixel Clk: 297000000Hz		TMDS Clk: 297000000Hz
Color Format: YUV422		Color Depth: 10 bit
Colorimetry: ITU.BT2020		EOTF: HLG
x0: 35400				y0: 14599
x1: 8500				y1: 39850
x2: 6550				y2: 2300
white x: 15635			white y: 16451
max lum: 0			min lum: 0
max cll: 0			max fall: 0
----
VOP [ff370000.vop]: ACTIVE
    Connector: HDMI-A
	overlay_mode[1] bus_format[2016] output_mode[f] color_space[10]
    Display mode: 3840x2160p30
	clk[297000] real_clk[297000] type[40] flag[5]
	H: 3840 4016 4104 4400
	V: 2160 2168 2178 2250
    win0-0: ACTIVE
	format: NA12 little-endian (0x3231414e) HDR[3] color_space[10]
	csc: y2r[0] r2r[0] r2y[0] csc mode[0]
	zpos: 0
	src: pos[0x0] rect[3840x2160]
	dst: pos[0x0] rect[3840x2160]
	buf[0]: addr: 0x000000001ca2a000 pitch: 4864 offset: 0
	buf[1]: addr: 0x000000001ca2a000 pitch: 4864 offset: 10506240
    win1-0: DISABLED
    win2-0: DISABLED
    post: sdr2hdr[0] hdr2sdr[0]
    pre : sdr2hdr[0]
    post CSC: r2y[0] y2r[0] CSC mode[1]
-----
             armclk                       0            0   816000000          0 0  
          sclk_vdec_core                  2            3   300000000          0 0  
             aclk_rkvdec                  3            4   600000000          0 0  
          sclk_vdec_cabac                 2            3   300000000          0 0  
          sclk_ddrc                       2            2   786000000          0 0  
```

LongChair, 2018-01-27:
@mcerveny : great ! and thanks for sharing once again :)
Would you mind summing up the different properties that are required then ?

mcerveny, 2018-01-27:
Check the code mpi_dec.c in [utils](https://github.com/mcerveny/utils):
* program initialization - choose [plane](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L678) (only plane [win0/primary](https://github.com/rockchip-linux/kernel/blob/15dbb651be03ed7f9905f2e7c54b61d55011f54a/drivers/gpu/drm/rockchip/rockchip_drm_vop.c#L711) is capable of all HDR in rk3328 (generally you should check plane formats (DRM_FORMAT_NV12, DRM_FORMAT_NV12_10) and "FEATURE" witch window support [HDR to SDR](https://github.com/rockchip-linux/kernel/blob/15dbb651be03ed7f9905f2e7c54b61d55011f54a/drivers/gpu/drm/rockchip/rockchip_vop_reg.c#L992)) and ["HDR_PANEL_METADATA"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L615)
* stream metadata decode - set all [parameters+scaling](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L379) 
   * plane ["COLOR_SPACE"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L401)
   * plane ["EOTF"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L405) 
   * if HDR output, set connector ["HDR_SOURCE_METADATA"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L407)
   * refresh connector chain ["hdmi_output_format"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L417)
* stream frame decode - change ["FB_ID"](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L510) only
* do not forget [reset properties](https://github.com/mcerveny/utils/blob/master/mpi_dec.c#L825) after stream/program ends

mcerveny, 2018-01-29:
I looked to [rockchip-linux](https://github.com/rockchip-linux/kernel/blob/15dbb651be03ed7f9905f2e7c54b61d55011f54a/drivers/gpu/drm/rockchip/rockchip_drm_vop.c#L722) code and many up-conversion (like SDR2HDR_FOR_HLG_HDR), down-conversion and whole cross-coversion code missing  (between different plane and connector EOTF>0 (PQ<->HLG<->LOG) check [slide 15](http://files.syabas.com/popcornhour/doc/4K-HDR-Benchmark-v2.pdf)). (also ignored by rockchip rockchip-linux/kernel#61).

mcerveny, 2018-01-31:
HDR playback resolution 3840x2160p60 is partially solved by mcerveny/rockchip-linux@9e1bae4461ea08ac668b7cc205ccdb83e17fd678. I am not able to decode/resolve problems in undocumented (and probably untested) [rockchip registry setting part](https://github.com/mcerveny/rockchip-linux/blob/9e1bae4461ea08ac668b7cc205ccdb83e17fd678/drivers/phy/rockchip/phy-rockchip-inno-hdmi-phy.c#L811) or similar.

Kwiboo, 2018-01-31:
@mcerveny nice findings 👍, I will test 2160p@60hz 10-bit 4:2:0 mode with your patch ASAP.

Agree, it would be great to have documentation for the inno hdmi phy. Some hw register information can be deduced from https://github.com/rockchip-linux/kernel/blob/release-3.10/drivers/video/rockchip/hdmi/rockchip-hdmiv2/rockchip_hdmiv2_hw.h#L1584-L1750, `EXT_PHY` and `EXT_PHY1` should mainly differ in register address.

There is also two inno hdmi phy related commits in the recent `release-4.4` push that might be of interest, see: https://github.com/rockchip-linux/kernel/commit/bc980e7ce24f1ad1f5f205a7be49fc1955570c02 and https://github.com/rockchip-linux/kernel/commit/fe62b85484dc701cf21908b23a31357d823b37cf

mcerveny, 2018-05-04:
4k60 HDR+HDR/HLG (10bit 4:2:0 - TMDS clock 371Mhz) with never [patched rockchip-4.4 kernel](https://github.com/mcerveny/rockchip-linux/tree/release-4.4-hdr-201805) is working (tested on LG TV). TMDS Clk: 92812500Hz in output is [OK](https://github.com/rockchip-linux/kernel/blob/5c831a95a12f355a9ba03bd6fd16c12f4174d74e/drivers/gpu/drm/bridge/synopsys/dw-hdmi.c#L3154-L3159). 
```
# ./mpi_dec ../4k60hdr.hevc 16777220 > /dev/null
# ( echo ---; cat /sys/kernel/debug/dw-hdmi/status; echo ----; cat /sys/kernel/debug/dri/0/summary; echo -----; egrep 'armclk |aclk_rkvdec |cabac |vdec_core |ddrc |hdmi' /sys/kernel/debug/clk/clk_summary; echo ----; cat /sys/class/thermal/thermal_zone0/temp )
---
PHY: enabled			Mode: HDMI
Pixel Clk: 594000000Hz		TMDS Clk: 92812500Hz
Color Format: YUV420		Color Depth: 10 bit
Colorimetry: ITU.BT2020		EOTF: ST2084
x0: 35400				y0: 14599
x1: 8500				y1: 39850
x2: 6550				y2: 2300
white x: 15635			white y: 16451
max lum: 0			min lum: 0
max cll: 0			max fall: 0
----
VOP [ff370000.vop]: ACTIVE
    Connector: HDMI-A
	overlay_mode[1] bus_format[2027] output_mode[e] color_space[10]
    Display mode: 3840x2160p60
	clk[594000] real_clk[594000] type[48] flag[5]
	H: 3840 4016 4104 4400
	V: 2160 2168 2178 2250
    win0-0: ACTIVE
	format: NA12 little-endian (0x3231414e) HDR[2] color_space[10]
	csc: y2r[0] r2r[0] r2y[0] csc mode[1]
	zpos: 0
	src: pos[0x0] rect[3840x2160]
	dst: pos[0x0] rect[3840x2160]
	buf[0]: addr: 0x000000001b11d000 pitch: 4864 offset: 0
	buf[1]: addr: 0x000000001b11d000 pitch: 4864 offset: 10506240
    win1-0: DISABLED
    win2-0: DISABLED
    post: sdr2hdr[0] hdr2sdr[0]
    pre : sdr2hdr[0]
    post CSC: r2y[0] y2r[0] CSC mode[1]
-----
    hdmi_phy                              1            1   594000000          0 0  
       hdmiphy                            1            1   594000000          0 0  
          hdmiphy_peri                    0            0   594000000          0 0  
    clk_hdmi_sfc                          1            1    24000000          0 0  
             armclk                       0            0  1008000000          0 0  
          sclk_vdec_core                  2            3   100000000          0 0  
             aclk_rkvdec                  3            4   100000000          0 0  
                   pclk_hdmiphy           1            1    75000000          0 0  
                pclk_hdmi                 1            1   100000000          0 0  
          sclk_vdec_cabac                 2            3   100000000          0 0  
             dclk_hdmiphy                 0            0     8000000          0 0  
          sclk_ddrc                       2            2   924000000          0 0  
----
62916
```

LongChair, 2018-05-04:
@mcerveny : nice :) we're based on an older kernel still, and HDR works, but not in 60 fps.
Any idea on which commit fixed that ?

mcerveny, 2018-05-04:
I did not make bisect. But https://github.com/rockchip-linux/kernel/commit/d4953cfed336fc3197ba8e98d69ea6a37f2b5fdc equals to my https://github.com/mcerveny/rockchip-linux/commit/b4bc703f2dca4e5115b22155920d2277671a9f00 and add something new for faster "clock ratio". Maybe more in history [sysnopsys](https://github.com/rockchip-linux/kernel/commits/release-4.4/drivers/gpu/drm/bridge/synopsys), [drm/rockchip](https://github.com/rockchip-linux/kernel/commits/release-4.4/drivers/gpu/drm/rockchip) and/or [phy/rockchip](https://github.com/rockchip-linux/kernel/commits/release-4.4/drivers/phy/rockchip).

mo123, 2018-06-19:
@mcerveny Hi, Can you rebase your changes to latest 4.4 kernel & mpp.
I want to test HDR on RK3328 LibreELEC, what changes are still needed for 4K@60hz HDR?

mcerveny, 2018-08-09:
@Kwiboo: 
Hi, I see that you moving forward (xbmc, ffmpeg+rkvpu). 
Did you successfully implemented HDR/HLG chain in MPV/KODI ? 

I have still performance issue with "LG Chess 4K Demo.mp4" in [mpi_dec tester](https://github.com/mcerveny/utils) or with ffmpeg. Even if I set cpu>=1008MHz and mem==924MHz with actual rockchip-4.4 kernel and mpp+ffpeg from github I am getting only 57 FPS eg. missing additional 5-7% performance in mpp to get playable output. Are you in contact with rockchip to resolve this problem (https://github.com/rockchip-linux/kernel/issues/59) ?

lorenz, 2018-08-09:
HDR10 (SMTPE ST 2084/BT2020/YUV420 10bit) works on the RK3328 with Kodi, but there are still files that break the decoder (colors all over the place/black) or where it can't keep up (mainly 60FPS content, but also everything over 120Mbps).

closed 2020-01-01 by Kwiboo