Kwiboo/linux-rockchip #24: RK3328 : color space & sdr/hdr path missing

Kwiboo, 2017-08-21:
The 4.4 kernel tree is missing csc / hdr2sdr code for rk3328 resulting in video colors looking bad compared to Android and the 3.10.104 kernel. Colorspace path appears to be implemented only for rk3399 in 4.4.

yanghanxing, 2017-08-23:
This feature is WIP. 
I will notice you to update once it completed.

mcerveny, 2018-01-26:
I think that it is working fine now. Example HDR->SDR: 
- input: win0 plane with NA12 (Y420-10bit), COLOR_SPACE=V4L2_COLORSPACE_BT2020, EOTF=SMPTE_ST2084
- plane_conversion: y2r[1] (YUV to RGB), CSC[3] is CSC_BT2020
- post_conversion: hdr2sdr[1], CSC[1] is CSC_BT709L
- output: RGB 8 bit
```
# cat /sys/kernel/debug/dri/0/summary
VOP [ff370000.vop]: ACTIVE
    Connector: HDMI-A
	overlay_mode[0] bus_format[100a] output_mode[f] color_space[0]
    Display mode: 1280x1024p60
	clk[108000] real_clk[108000] type[48] flag[5]
	H: 1280 1328 1440 1688
	V: 1024 1025 1028 1066
    win0-0: ACTIVE
	format: NA12 little-endian (0x3231414e) HDR[1] color_space[10]
	csc: y2r[1] r2r[0] r2y[0] csc mode[3]
	zpos: 0
	src: pos[0x0] rect[3840x2160]
	dst: pos[0x152] rect[1280x720]
	buf[0]: addr: 0x0000000001eb6000 pitch: 4864 offset: 0
	buf[1]: addr: 0x0000000001eb6000 pitch: 4864 offset: 10506240
    win1-0: DISABLED
    win2-0: DISABLED
    post: sdr2hdr[0] hdr2sdr[1]
    pre : sdr2hdr[0]
    post CSC: r2y[0] y2r[0] CSC mode[1]
# cat /sys/kernel/debug/dw-hdmi/status 
PHY: enabled			Mode: DVI
Pixel Clk: 108000000Hz		TMDS Clk: 108000000Hz
Color Format: RGB		Color Depth: 8 bit
Colorimetry: ITU.BT709		EOTF: Off
```
For more examples (HDR->SDR->HDR) and (HDR->passthrough->HDR) check [here](https://github.com/Kwiboo/linux-rockchip/issues/22#issuecomment-360531385).

closed 2020-01-01 by Kwiboo