Kwiboo/linux-rockchip #25: HD-Audio passthrough settings not present

infinity85, 2017-08-28:
Hi there hardworking early bird developers :D

Sorry if I'm wrong by writing here. Not sure whether it is tolerated to mention kind of standard issues at the this stage of development. So sorry if this is inappropriate, just give me a hint then :)

I recently got my ROCK64 (4GB) board and burned the last rockchip-alpha4 onto an µSD-Card. 
Auto frame rate switching already works appartently as well as GigabitLAN.

Though, I'd like to mention that DTS-HD and TrueHD passthrough settings are missing (expert settings-view enabled). 

There are a lot of other things/issues I'd like to mention after the first test, but I don't know where to do it, because I think the best would be an open discussion at forum.libreelec.tv or forum.kodi.tv, but I didn't see any threads from you guys?

Really appreciate your effort and the great progress!

Kwiboo, 2017-08-28:
Hi there and thanks! :)

Finalizing proper NL-PCM and HBR audio support is on my agenda for this week and will require mapping the 16-bit samples into a 21-bit sample before sending it to the i2s fifo.
AC3, E-AC3 and DTS is currently semi-working because TV/AVR's detects the incoming LPCM audio as compressed audio.

The options for DTS-HD and TrueHD should still be shown in the menu, the output from `hexdump -C /sys/class/drm/card0-HDMI-A-1/edid` should tell if there is something missing in the video/audio detection.

It is a little bit too early for reporting general issues, the current state is more of a technical proof-of-concept, once the audio work has completed next step will be to rebase on top of LE/Kodi master.

The Pine64 IRC channel for ROCK64 is probably the easiest and fastest way for quick discussion at this stage.

LongChair, 2017-09-03:
Closing this for now :)

closed 2017-09-03 by LongChair