Kwiboo/linux-rockchip #28: What command for VPU ioctl

dalmatele, 2017-10-03:
I want to call Encode function of VPU.
Can you show me the value of Command for ioctl?
As the kernel, I see this:
```
#define MPP_IOC_MAGIC			'l'

#define MPP_IOC_SET_CLIENT_TYPE		_IOW(MPP_IOC_MAGIC, 1, u32)
#define MPP_IOC_GET_HW_FUSE_STATUS	_IOW(MPP_IOC_MAGIC, 2, u32)

#define MPP_IOC_SET_REG			_IOW(MPP_IOC_MAGIC, 3, u32)
#define MPP_IOC_GET_REG			_IOW(MPP_IOC_MAGIC, 4, u32)

#define MPP_IOC_PROBE_IOMMU_STATUS	_IOR(MPP_IOC_MAGIC, 5, u32)
```
But when I call ioctl with this:

```
#define VPU_IOC_SET_CLIENT_TYPE             _IOW(VPU_IOC_MAGIC, 1, __u32)
#define VPU_IOC_GET_HW_FUSE_STATUS          _IOW(VPU_IOC_MAGIC, 2, __u32)
#define VPU_IOC_SET_REG                     _IOW(VPU_IOC_MAGIC, 3, __u32)
#define VPU_IOC_GET_REG                     _IOW(VPU_IOC_MAGIC, 4, __u32)
#define VPU_IOC_PROBE_IOMMU_STATUS          _IOR(VPU_IOC_MAGIC, 5, __u32)
```
The kernel is crashed (it hangs and we must restart device).

Kwiboo, 2017-10-03:
We have only tested decoding for LibreELEC usage.

What kernel and patches are you using? I recommend you to test with/without the pl330 topic branch as that solved some dma related lockups we have with decoding.

Kwiboo, 2017-10-08:
Closing issue, please create an issue in upstream https://github.com/rockchip-linux/kernel and/or https://github.com/rockchip-linux/mpp.

closed 2017-10-08 by Kwiboo