Kwiboo/linux-rockchip #30: rk3399 libhybris and Android gralloc Mali driver

ovidiu31, 2017-10-19:
Hello @Kwiboo 

Would be possible to add support in your build environment for ARM Mali-T860 MP4 GPU from the rk3399 SoC?
I have working kernel 4.4.55 #474 SMP Tue Sep 5 09:09:33 CST 2017 aarch64 GNU/Linux bootloader and almost all the stuff compiling and running fine with your builds except the GPU/VPU acceleration.
Thank you for reading and considering.

Kwiboo, 2017-10-19:
Hello @ovidiu31, I only have rk3288 (ASUS Tinker Board) and rk3328 (ROCK64, Z28 and soon RockBox) devices so I cannot test or verify if anything works on rk3399 or not.

The `mali-rockchip` package should already support installing Mali-T860 user-space drivers (untested).
I think that if you create a new `projects/Rockchip/devices/$DEVICE` folder, changes `$DEVICE/options` to use `MALI_FAMILY="t860"` and uses a custom `$DEVICE/linux/linux.aarch64.conf` tailored for your device/rk3399 it should be possible.

ovidiu31, 2017-10-19:
thanks a lot for help and assist.
if you want i can send you an development board with rk3399

ovidiu31, 2017-10-23:
@Kwiboo 
failed to compile for rk3399 with the modifications you suggested

make[3]: Leaving directory '/root/LibreELEC.tv/build.LibreELEC-RK3399.aarch64-8.2-devel/cmake-3.6.3/.x86_64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:1729: Utilities/cmlibarchive/libarchive/CMakeFiles/cmlibarchive.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 47%] Linking CXX static library libcmsys.a
make[3]: Leaving directory '/root/LibreELEC.tv/build.LibreELEC-RK3399.aarch64-8.2-devel/cmake-3.6.3/.x86_64-linux-gnu'
[ 47%] Built target cmsys
[ 48%] Linking C static library libcmcurl.a
make[3]: Leaving directory '/root/LibreELEC.tv/build.LibreELEC-RK3399.aarch64-8.2-devel/cmake-3.6.3/.x86_64-linux-gnu'
[ 48%] Built target cmcurl
make[2]: Leaving directory '/root/LibreELEC.tv/build.LibreELEC-RK3399.aarch64-8.2-devel/cmake-3.6.3/.x86_64-linux-gnu'
make[1]: *** [Makefile:161: all] Error 2
make[1]: Leaving directory '/root/LibreELEC.tv/build.LibreELEC-RK3399.aarch64-8.2-devel/cmake-3.6.3/.x86_64-linux-gnu'
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 2
root@diabloss:~/LibreELEC.tv#

closed 2020-01-01 by Kwiboo