Kwiboo/linux-rockchip #35: Some incorrectly decoded H.264 videos

sigmaris, 2020-11-02:
Hi @Kwiboo,

Thanks for all your work on the rkvdec and hantro video decoders in the kernel and FFmpeg. Thanks to you I have been able to assemble a very capable media center using a Rockpro64 board and Kodi. It plays most videos very well, including 10-bit with full hardware acceleration :) . However I have found a few H.264 videos which show some artifacts / corruption when decoded with rkvdec, here are a couple of samples:

https://sigmaris.info/files/titles1.mkv
https://sigmaris.info/files/titles2.mkv

These two short clips are from movies and when played on Kodi the picture is mostly recognisable but there are a lot of blocky artifacts in the picture that look wrong (slightly wrong colours, or mixed with other nearby parts of the picture).

I've also noticed similar issues on the terrestrial HDTV broadcasts here in the UK, which are encoded in H.264. The corruption is similar but seems to get worse/better in cycles every 1-2 seconds, I guess this is related to the interval of I-frames being encoded in the broadcast. Here is a sample of HDTV:

https://sigmaris.info/files/bbc_1_hd.ts

Most other H.264 videos, including everything I've viewed on Youtube and Twitch live streams look 100% OK. I am mainly uploading these samples in case it helps finding bugs in the rkvdec driver or FFmpeg.

I am using this kernel based on 5.9.3 plus your branch: https://github.com/sigmaris/linux/releases/tag/5.9.3-rockpro64-ci
and this FFmpeg: https://github.com/sigmaris/FFmpeg/releases/tag/4.3.1-25_sigmaris
I am using the rkvdec driver to decode H.264, I haven't enabled H.264 decoding in the RK3399 hantro driver.

sigmaris, 2021-02-06:
Update - using the recently updated patches from LibreELEC master, including https://github.com/LibreELEC/LibreELEC.tv/pull/4988 I've got a matched kernel API and FFmpeg which decodes all the H.264 videos properly, including the samples here which previously caused problems.

hptsf, 2021-09-22:
hello，
What is the fps you tested?
and what's the command which you using？

closed 2026-08-18 by sigmaris