Kwiboo/linux-rockchip #38: [rk3528][NanoPi Zero2] rcu_preempt detected stalls on CPUs/tasks

drozdi70, 2025-07-18:
While starting new kernel on NanoPi Zero2 I see:

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.16.0-rc5-gd7b8f8e20813-dirty (orangepi@NucBoxK6) (aarch64-linux-gnu-gcc (ctng-1.25.0-119g-FA) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Fri Jul 18 14:16:54 CEST 2025
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: FriendlyElec NanoPi Zero2
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff9f0000 (options '')
[    0.000000] printk: legacy bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] OF: reserved mem: 0x000000000010f000..0x000000000010f0ff (0 KiB) nomap non-reusable shmem@10f000
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x7fbd6a40-0x7fbd90bf]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000083fffff]
[    0.000000]   node   0: [mem 0x0000000008c00000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 2048 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x000000007ba00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 25 pages/cpu s62232 r8192 d31976 u102400
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: storagemedia=sd androidboot.storagemedia=sd androidboot.mode=normal androidboot.verifiedbootstate=orange earlycon=uart8250,mmio32,0xff9f0000 console=ttyFIQ0 coherent_pool=1m rw root=/dev/mmcblk2p8 rootfstype=ext4 rootflags=discard #o-07/18/2025 androidboot.fwver=ddr-v1.10-44ec2e0a51,spl-v1.05,bl31-v1.18,bl32-v1.06,uboot-9dc8987
[    0.000000] Unknown kernel command line parameters "#o-07/18/2025 storagemedia=sd", will be passed to user space.
[    0.000000] printk: log buffer data + meta data: 131072 + 458752 = 589824 bytes
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007b600000-0x000000007b800000] (2MB)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521728
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=512 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001637] Console: colour dummy device 80x25
[    0.002130] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003098] pid_max: default: 32768 minimum: 301
[    0.003592] LSM: initializing lsm=capability
[    0.004098] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.004777] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 212 lines cut ...]
[  420.213177] x2 : 000000000004376c x1 : 0000000000000001 x0 : 4000000000000000
[  420.213830] Call trace:
[  420.214054]  default_idle_call+0x28/0x3c (P)
[  420.214447]  do_idle+0x200/0x25c
[  420.214748]  cpu_startup_entry+0x38/0x3c
[  420.215111]  secondary_start_kernel+0x138/0x158
[  420.215529]  __secondary_switched+0xc0/0xc4
[  483.232329] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  483.232882] rcu:     2-...0: (1 GPs behind) idle=0654/1/0x4000000000000000 softirq=43/44 fqs=20634
[  483.233673] rcu:     (detected by 1, t=120730 jiffies, g=-1131, q=147 ncpus=4)
[  483.234303] Sending NMI from CPU 1 to CPUs 2:
[  546.272329] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  546.272881] rcu:     2-...0: (1 GPs behind) idle=0654/1/0x4000000000000000 softirq=43/44 fqs=23287
[  546.273672] rcu:     (detected by 1, t=136490 jiffies, g=-1131, q=147 ncpus=4)
[  546.274303] Sending NMI from CPU 1 to CPUs 2:
[  609.312329] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  609.312881] rcu:     2-...0: (1 GPs behind) idle=0654/1/0x4000000000000000 softirq=43/44 fqs=25940
[  609.313671] rcu:     (detected by 1, t=152250 jiffies, g=-1131, q=147 ncpus=4)
[  609.314302] Sending NMI from CPU 1 to CPUs 2:
[  672.352328] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  672.352882] rcu:     2-...0: (1 GPs behind) idle=0654/1/0x4000000000000000 softirq=43/44 fqs=28592
[  672.353673] rcu:     (detected by 3, t=168010 jiffies, g=-1131, q=147 ncpus=4)
[  672.354304] Sending NMI from CPU 3 to CPUs 2:

Using DTS tree taken from Offcial https://github.com/friendlyarm/kernel-rockchip --depth 1 -b nanopi6-v6.1.y kernel-rk3528-friendlyarm

drozdi70, 2025-07-20:
decode:

[   21.288313] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[   21.288877] rcu:     0-...0: (5 ticks this GP) idle=04cc/1/0x4000000000000000 softirq=38/44 fqs=2579
[   21.289682] rcu:     (detected by 3, t=5252 jiffies, g=-1139, q=146 ncpus=4)
[   21.290298] Sending NMI from CPU 3 to CPUs 0:
[   31.291558] rcu: rcu_preempt kthread timer wakeup didn't happen for 2501 jiffies! g-1139 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402
[   31.292973] rcu:     Possible timer handling issue on cpu=3 timer-softirq=2591
[   31.293601] rcu: rcu_preempt kthread starved for 2503 jiffies! g-1139 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[   31.294540] rcu:     Unless rcu_preempt kthread gets sufficient CPU time, OOM is now expected behavior.
[   31.295360] rcu: RCU grace-period kthread stack dump:
[   31.295816] task:rcu_preempt     state:I stack:0     pid:15    tgid:15    ppid:2      task_flags:0x208040 flags:0x00000010
[   31.296817] Call trace:
[   31.297041] __switch_to (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/arch/arm64/kernel/process.c:740) (T)
[   31.297402] __schedule (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/core.c:5404 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/
kernel-rk3528/kernel/sched/core.c:6790)
[   31.297725] schedule (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./arch/arm64/include/asm/preempt.h:54 /home/orangepi/work/Z2/test_new_dts/sd-f
use_rk3528/kernel-rk3528/kernel/sched/core.c:6869 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/core.c:6883)
[   31.298025] schedule_timeout (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/sleep_timeout.c:100)
[   31.298378] rcu_gp_fqs_loop (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/rcu/tree.c:2054 (discriminator 13))
[   31.298742] rcu_gp_kthread (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/rcu/tree.c:2259)
[   31.299096] kthread (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/kthread.c:464)
[   31.299394] ret_from_fork (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/arch/arm64/kernel/entry.S:863)
[   31.299726] rcu: Stack dump where RCU GP kthread last ran:
[   31.300223] CPU: 3 UID: 0 PID: 0 Comm: swapper/3 Not tainted 6.16.0-rc5-gd7b8f8e20813-dirty #1 PREEMPT
[   31.301071] Hardware name: FriendlyElec NanoPi Zero2 (DT)
[   31.301559] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   31.302189] pc : default_idle_call (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/idle.c:126)
[   31.302580] lr : default_idle_call (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./arch/arm64/include/asm/irqflags.h:26 /home/orangepi/work/Z2/te
st_new_dts/sd-fuse_rk3528/kernel-rk3528/./arch/arm64/include/asm/irqflags.h:48 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/idle.c:124)

[   31.302970] sp : ffff800082f9be00
[   31.303270] x29: ffff800082f9be00 x28: 0000000000000000 x27: 0000000000000000
[   31.303922] x26: 0000000000000000 x25: ffff00000355e900 x24: 0000000000000000
[   31.304574] x23: 0000000000000000 x22: ffff8000828eafb0 x21: ffff8000820140b0
[   31.305225] x20: ffff8000828eae70 x19: 0000000000000003 x18: 0000000000000000
[   31.305877] x17: 000000000000001c x16: 000000004616d7d8 x15: ffff00007fbb5980
[   31.306528] x14: 00000000000002fc x13: 0000000000000001 x12: 0000000000000000
[   31.307179] x11: 0000000000000001 x10: 0000000000000ab0 x9 : ffff800082f9bd50
[   31.307831] x8 : ffff00000355f410 x7 : 00000000000000c0 x6 : ffff800082011108
[   31.308482] x5 : ffff00007fbb1108 x4 : ffff800082011118 x3 : 0000000000000000
[   31.309133] x2 : 000000000000aa84 x1 : 0000000000000001 x0 : 4000000000000000
[   31.309784] Call trace:
[   31.310007] default_idle_call (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/idle.c:126) (P)
[   31.310400] do_idle (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/idle.c:186 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kern
el-rk3528/kernel/sched/idle.c:325)
[   31.310701] cpu_startup_entry (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/sched/idle.c:422 (discriminator 1))
[   31.311062] secondary_start_kernel (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./arch/arm64/include/asm/atomic_ll_sc.h:95 /home/orangepi/work/Z
2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./arch/arm64/include/asm/atomic.h:28 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linux/atomic
/atomic-arch-fallback.h:546 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linux/atomic/atomic-arch-fallback.h:994 /home/orangepi/work/Z2/te
st_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linux/atomic/atomic-instrumented.h:436 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linu
x/sched/mm.h:37 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/arch/arm64/kernel/smp.c:214)
[   31.311479] __secondary_switched (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/arch/arm64/kernel/head.S:405)
[   94.328311] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[   94.328863] rcu:     0-...0: (5 ticks this GP) idle=04cc/1/0x4000000000000000 softirq=38/44 fqs=10317
[   94.329674] rcu:     (detected by 2, t=23512 jiffies, g=-1139, q=146 ncpus=4)
[   94.330295] Sending NMI from CPU 2 to CPUs 0:
[  157.368309] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  157.368860] rcu:     0-...0: (5 ticks this GP) idle=04cc/1/0x4000000000000000 softirq=38/44 fqs=16847
[  157.369671] rcu:     (detected by 3, t=39272 jiffies, g=-1139, q=146 ncpus=4)

drozdi70, 2025-07-30:
some steps forward.... now the issue with power domain as below:

[    0.094929] rockchip-pm-domain ff600000.power-management:power-controller: power-domain: failed to get clk at index 0: -517
[    0.095942] rockchip-pm-domain ff600000.power-management:power-controller: failed to handle node power-domain: -517

[    0.567762] pwm-regulator vdd-cpu: error -EBUSY: Failed to get PWM
[    0.568350] pwm-regulator vdd-cpu: probe with driver pwm-regulator failed with error -16

drozdi70, 2025-08-10:
next issue while mounting rootfs:

`[    0.666184] clk: Disabling unused clocks
[    0.667218] dw-apb-uart ff9f0000.serial: forbid DMA for kernel console
[    0.676867] ehci-platform ff100000.usb: USB 2.0 started, EHCI 1.00
[    0.676935] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    0.678128] hub 1-0:1.0: USB hub found
[    0.678613] hub 1-0:1.0: 1 port detected
[    0.692899] mmc2: SDHCI controller on ffbf0000.mmc [ffbf0000.mmc] using ADMA
[    0.721475] hub 2-0:1.0: USB hub found
[    0.721852] hub 2-0:1.0: 1 port detected
[    0.727278] Freeing unused kernel memory: 11456K
[    0.727820] Run /init as init process
Loading, please wait...
Starting systemd-udevd version 255.4-1ubuntu8.4
[    0.798264] mmc2: new HS200 MMC card at address 0001
[    0.801530] mmcblk2: mmc2:0001 A3V014 233 GiB
[    0.806108]  mmcblk2: p1 p2 p3 p4 p5 p6 p7 p8 p9
[    0.808415] mmcblk2boot0: mmc2:0001 A3V014 4.00 MiB
[    0.813557] mmcblk2boot1: mmc2:0001 A3V014 4.00 MiB
[    0.816184] mmcblk2rpmb: mmc2:0001 A3V014 16.0 MiB, chardev (511:0)
[    0.872889] mmc_host mmc0: Bus speed (slot 0) = 148500000Hz (slot req 150000000Hz, actual 148500000HZ div = 0)
[    0.888102] dwmmc_rockchip ffc30000.mmc: Successfully tuned phase to 276
[    0.888794] mmc0: new UHS-I speed SDR104 SDXC card at address aaaa
[    0.890395] mmcblk0: mmc0:aaaa SC64G 59.5 GiB
[    0.898255]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
[    2.468845] Unable to handle kernel write to read-only memory at virtual address ffff80008163b2a8
[    2.469658] Mem abort info:
[    2.469910]   ESR = 0x000000009600004e
[    2.470244]   EC = 0x25: DABT (current EL), IL = 32 bits
[    2.470717]   SET = 0, FnV = 0
[    2.470991]   EA = 0, S1PTW = 0
[    2.471273]   FSC = 0x0e: level 2 permission fault
[    2.471700] Data abort info:
[    2.471958]   ISV = 0, ISS = 0x0000004e, ISS2 = 0x00000000
[    2.472445]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[    2.472893]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    2.473365] swapper pgtable: 4k pages, 48-bit VAs, pgdp=00000000021c7000
[    2.473957] [ffff80008163b2a8] pgd=0000000000000000, p4d=10000000032ed003, pud=10000000032ee003, pmd=0060000001a00781
[    2.474905] Internal error: Oops: 000000009600004e [#1]  SMP
[    2.475408] Modules linked in:
[    2.475689] CPU: 1 UID: 0 PID: 0 Comm: swapper/1 Not tainted 6.16.0-rc7-next-20250722-g8e0a28d375ed-dirty #1 PREEMPT
[    2.476626] Hardware name: FriendlyElec NanoPi Zero2 (DT)
[    2.477106] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    2.477723] pc : __run_timers (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linux/list.h:982 /home/orangepi/work/Z2/test_new_dts/sd-fus
e_rk3528/kernel-rk3528/kernel/time/timer.c:891 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:1781 /home/orangepi/work/Z2/test_new
_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:2372)
[    2.478091] lr : __run_timers (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:2371)
[    2.478451] sp : ffff80008000bd20
[    2.478749] x29: ffff80008000bd20 x28: ffff00000355c600 x27: ffff80008000bd80
[    2.479388] x26: ffff8000829089c0 x25: ffff00007fb8fb68 x24: ffff80008000bd80
[    2.480028] x23: 0000000000000000 x22: dead000000000122 x21: ffff80008000bef0
[    2.480666] x20: ffff00007fb8fb40 x19: ffff0000050f29b0 x18: 0000000000000000
[    2.481305] x17: ffff7ffffdb3b000 x16: ffff800080008000 x15: 0000000000000000
[    2.481944] x14: 00000d4d116de9ca x13: 0000000000000000 x12: 0000000000000089
[    2.482584] x11: ffff00007fb66b00 x10: 0000000000000305 x9 : ffff00007fb8fbb0
[    2.483222] x8 : 0000000000000001 x7 : ffff80008000bd88 x6 : 000000000000006e
[    2.483860] x5 : 0010000000000020 x4 : 0000000000000240 x3 : 0000000000000009
[    2.484497] x2 : 0000000000000200 x1 : ffff80008000bd80 x0 : ffff80008163b2a0
[    2.485136] Call trace:
[    2.485357] __run_timers (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/./include/linux/list.h:982 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3
528/kernel-rk3528/kernel/time/timer.c:891 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:1781 /home/orangepi/work/Z2/test_new_dts/
sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:2372) (P)
[    2.485720] timer_expire_remote (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:2385 /home/orangepi/work/Z2/test_new_dts/sd-fus
e_rk3528/kernel-rk3528/kernel/time/timer.c:2376 /home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer.c:2135)
[    2.486091] tmigr_handle_remote_up (/home/orangepi/work/Z2/test_new_dts/sd-fuse_rk3528/kernel-rk3528/kernel/time/timer_migration.c:944 /home/orangepi/work/Z2/test_ne
[... 28 lines cut ...]
[ 2.492825] Code: f9000693 a9400660 f9000020 b4000040 (f9000401)
All code
========
   0:*  93                      xchg   %eax,%ebx                <-- trapping instruction
   1:   06                      (bad)
   2:   00 f9                   add    %bh,%cl
   4:   60                      (bad)
   5:   06                      (bad)
   6:   40 a9 20 00 00 f9       rex test $0xf9000020,%eax
[... cut ...]
===========================================
   0:   01 04 00                add    %eax,(%rax,%rax,1)
   3:   f9                      stc
[    2.493369] ---[ end trace 0000000000000000 ]---
[    2.493781] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[    2.494389] SMP: stopping secondary CPUs
[    2.494747] Kernel Offset: disabled
[    2.495058] CPU features: 0x00000,00040000,10002000,0400421b
[    2.495560] Memory Limit: none
[    2.495838] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---`

Kwiboo, 2025-08-10:
@drozdi70 please build your Linux kernel with patches from https://github.com/Kwiboo/linux-rockchip/commits/next-20250722-rk3528/ and boot using U-Boot from https://source.denx.de/u-boot/contributors/kwiboo/u-boot/-/commits/rk3528 until more patches has landed upstream.

Using a downstream vendor device tree on mainline Linux is not expected to fully work, my tree have a device tree for the NanoPi Zero2 that works without issues on my board.